eriknn/ha-pax_ble #1: Two fans are not working

aen42, 2022-08-18:
First, a big thank you for publishing this code!

it looks like it is working great if I attach one Pax Calima. But when I configure the second fan no more updates happens on the first fan. Restarting HA does not help. 

In the configuration screen I get two lines in the box with Pax calima integration. The first line has two units (both fans). The second line has only one. I think this is what is causing one of the fans to become unavailable and likely what prevents updates on the other. 

What can I do to debug this further? I do know some python but am totally new to HomeAssistant…

eriknn, 2022-08-18:
Hey, didn’t know anyone would find my project so soon - this is my first hithub project.
The behaviour is expected as I haven’t completely fixed this part - it is due to the second coordinator overwriting the first in init.py. I’ll have a look at it when I have time.

I am also struggling with the HA bluetooth integration some times requiring complete restart, but this I don’t think is because of my code.

eriknn, 2022-08-18:
Try it now. I don't have two fans so I can't really test it, but debug shows that it tries to read from one that I added (with a different       MAC address). Also I removed the configuration option as it just messed everything up by adding new entities without removing the old.

aen42, 2022-08-19:
Thanks for the quick update!

I have just updated and restarted and so far it looks good! I don't mind the two config lines. If I see any more issue I will let you know. But thanks a lot for the quick fix!

closed 2022-08-19 by aen42