eriknn/ha-pax_ble #3: Connection failed

lazi3b0y, 2022-10-07:
Hello there :)

To be honest I'm not entirely sure if its this integration that is the problem, but I figured I'd start here..
Anyway, it seems like I'm unable to integrate my PAX fans after the latest home assistant update 2022.10.
The only thing I'm getting it "Connection failed".

eriknn, 2022-10-07:
I know, I have the same issue. I have it with other BT integrations as well, assuming it’s a bug in the HA update!

lazi3b0y, 2022-10-07:
I logged the exception that is thrown and got this:

'HaBleakScannerWrapper' object has no attribute 'find_device_by_address'

Doesn't tell me much since I'm not very used to python. I'll just leave it here, in case it helps in any way :)

eriknn, 2022-10-08:
Yeah, I checked and found the same! There’s been a breaking change there, so I’ll update my code when I find time!

eriknn, 2022-10-08:
Try it now!

I've had so much issues with this way to connect, and now realized the mac should be passed uppercase and not lowercase...

lazi3b0y, 2022-10-08:
Ah cool will try again tonight! :)

closed 2022-10-08 by lazi3b0y

lazi3b0y, 2022-10-08:
Seems to be working again, nice! :D