eriknn/ha-pax_ble #4: Two fans broke again

aen42, 2022-10-11:
Hi!
First: thanks for making this integration available!

My fans stopped working after update of home assistant to 2022.10.2. 

I then updated this integration and both fans were discovered as new (but I got four new entries). When I have configured one fan. The other will get errors when I try to connect it as well. Any idea on how this broke?

eriknn, 2022-10-11:
I think I made some changes to the setup. Did you try deleting them all and adding from scratch?

aen42, 2022-10-11:
Yes. I have tried to remove them and add again. Then at least one is working 😁

aen42, 2022-10-11:
Yay! I Added the second fan again manually using capitals in the mac and now it seems both fans are working again. However when I tried to add the second fan I see two pax Bluetooth integrations. Also I get a discover event for the same fan I just added. Perhaps I somehow have duplicate repos in hacs?!

eriknn, 2022-10-11:
Great! I don’t think capitals or not matters there, but adding manually probably helped.
I think the integration had a different name before, so you should definitely check if you have two integrations loaded!

Mvh
Erik Næss

eriknn, 2022-10-13:
Did you find anything?

aen42, 2022-10-13:
Sorry I have not tried again. The two fans are still working but I had to dismiss the superfluous discovered fan (same mac as the one I added manually). I do not have time to test this for quite some time in the future (likely until sometime in december). If you'd like we can close this for now and I can get back later if the issue persists.

eriknn, 2022-10-13:
Okay, manually vs discovered should really be the same things unless you have multiple integrations loaded, which is probably your issue. Will close it for now!

closed 2022-10-13 by eriknn

aen42, 2022-10-20:
I found there were three integrations installed and after I removed the two oldest I have not seen any new discoveries. And both fans seem to be working fine. Thanks for your help and for this integration!