eriknn/ha-pax_ble #5: 0% humidity

lazi3b0y, 2022-11-26:
My fans have started reporting 0% humidity whenever they aren't in boost mode. Anything anyone else have been experiencing? Both my fans started doing this at the same time it seems so that's why I suspect software issues.

eriknn, 2022-11-26:
Yeah, it’s the way it is in the library I’m using. It seems that the sensor is useless when the humidity is less than 40-ish %, and in general it shouldn’t be view as accurate.
Bote that there is no display of the humidity in the app either!

lazi3b0y, 2022-11-28:
It was working just fine earlier for me, just recently this has started happening, but you mean to say that you've had this issue all along?
Yeah, the Pax android app reset itself and is now crashing for me whenever I try to readd my fans to it, complete garbage sadly.

coolbombom, 2023-01-15:
I just bought an pax calima, and I'm experiencing the same problem. and the temperature sensor/reading is also off by 10 degrees. bummer :-(

eriknn, 2023-01-15:
lazi3boy: Yes, I,ve always had this issue. We read a number from the device, and this is converted to a humidity using a formula. When the read value is < something we set the humidity to 0. I assume this is something that makes sense.

coolbombom: Might be new FW that behaves differently? What is read va expected value? How/where is the fan mounted?

camelen, 2023-01-17:
This seems to be done on purpose. When the Calima shows below 30% it seems to automatically change to 0%.
Why is that?
This is a change from the Pycalima source.

Anyway I fixed it for myself.

Line 193 in Calima.py says:
        return FanState(round(math.log2(v[0]-30)*10, 2) if v[0] > 30 else 0, v[1]/4 - 2.6, v[2], v[3], trigger)

Compared to Pycalima which has this line below.
        return FanState(round(math.log2(v[0])*10, 2) if v[0] > 0 else 0, v[1]/4, v[2], v[3], trigger)

So if I change the v[0] argument to match the one below it seems to work (only tested for a few minutes, but displays similar result to ESP8266 with DHT22). Honestly it seems as if it also adjusts the temperature with -2.6 which also seems very strange to me. I'm also removing that which also more corresponds to my DHT22.

eriknn, 2023-01-17:
Yeah, this is probably correct - v[0] is humidity and v[1] is temp. I have corrected these to match my sensors. If this works better for you then it should maybe be changed back. Give it some time and check if it’s really better like this!

eriknn, 2023-02-22:
Any thoughts here? Is it better for tou guts without the correction? For me, it matches another sensor best with the correction…

Nakle, 2023-11-06:
@eriknn Same issue as @coolbombom here! Temperature is about 10 ish degrees off, and with me the humidity sensor is always zero when the fan is not spinning. Once it changes state, the humidity sensor seems correct.

eriknn, 2023-11-07:
Please do as follows:

Enable debugging for the PAX integration
For each read you should get a line with the following:
Read Fan States: (v[0], v[1], v[2], v[3], v[4], v[5], v[6])
Where those v-values are the raw data from the device.

- v[0] = Humidity
- v[1]  = Temp
- v[2] = Light
- v[3] = Fan speed
- v[4] = Mode (Actually "CurrentTrigger")
- v[5] = Tbd
- v[6] = Tbd

You'll see that the fan speed value for instance, will be correct.
Humidity and temp however, are not directly readable, and have to be converted somehow.
In the code which mine is built on, this was done as follows:

Humidity = round(math.log2(v[0])*10, 2) if v[0] > 0 else 0,
Temperature = v[1]/4

I am not able to find anything on this conversion in the Pax apk for Android. This is probably because they never display the values... Thus I'm not sure if this is actually correct.
Please feel free to look at the raw data in different situations and suggest better conversion.

closed 2024-02-14 by eriknn