eriknn/ha-pax_ble #6: Unable to connect to the fan

el97, 2023-01-06:
When I try to connect I get "Connection failed" as soon as I click save. I had my pi only a few meters away so I don't think its a distance issue. I use the 2023.1.0 version. There isn't much in the log but this is what I am getting:
```
2023-01-06 01:36:58.638 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration pax_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-06 01:53:49.795 INFO (MainThread) [custom_components.pax_ble.Calima] Not able to connect to <MAC-address>
```

eriknn, 2023-01-06:
I have no issues with 2023.1.1. Did you add manually or by discovery? If you debug the bluetooth driver, do you see it discovered?

el97, 2023-01-06:
I did some more testing and put my Pi close to the fan and surprisingly it started working. When I put it back in the original spot I can still see the fan when I use the bluetoothctl tool in terminal but it connects and disconnects frequently. Do the Pi need to be that close?
```
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> RSSI: -78
[CHG] Device <MAC-address> RSSI: -86
[CHG] Device <MAC-address> RSSI: -77
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -79
[CHG] Device <MAC-address> RSSI: -88
[CHG] Device <MAC-address> RSSI: -79
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -79
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -77
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -79
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -78
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -78
[CHG] Device <MAC-address> RSSI: -86
[CHG] Device <MAC-address> RSSI: -78
[CHG] Device <MAC-address> RSSI: -87
[CHG] Device <MAC-address> RSSI: -78
[CHG] Device <MAC-address> RSSI: -93
[CHG] Device <MAC-address> RSSI: -82
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> RSSI: -64
[CHG] Device <MAC-address> RSSI: -84
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> ServicesResolved: yes
[CHG] Device <MAC-address> ServicesResolved: no
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no
[CHG] Device <MAC-address> Connected: yes
[CHG] Device <MAC-address> Connected: no

eriknn, 2023-01-06:
I am using a NUC, so I’m not sure about the BT strength on the PI. I’m sure this is discussed elsewhere though.

el97, 2023-01-06:
Oh I see. Would it be to difficult to make it work with the proxy function for Bluetooth?

eriknn, 2023-01-06:
I didn’t even know that BT could be proxied. But I guess this is done on the BT-driver level, and would not require anything from the individual HASS-components.
Do you have more knowledge/documentation (links?) on this topic?

el97, 2023-01-06:
Here is some info together with ESPHome https://esphome.io/components/bluetooth_proxy
You can also take a look at this documentation https://www.home-assistant.io/integrations/bluetooth#remote-adapters

eriknn, 2023-01-06:
Looks like it should just work I think!

el97, 2023-01-07:
Do you mean as it is or do you need to do some changes in the code first?

eriknn, 2023-01-07:
I think it should just work as it is!

el97, 2023-01-07:
After a few tries I made it work. A bit strange since I can use a ESP module that is further away from the fan than the Pi itself.

closed 2023-01-07 by el97