eriknn/ha-pax_ble #7: Turn off autodiscovery?

jimmyl88, 2023-02-08:
Hi,
Thx for a nice intergration, i have added my 2 fans but "Pax Bluetooth" is scanning and offer me to configure 30+ new devices (not pax Calima devices).

How do i turn off autodiscovery?

eriknn, 2023-02-08:
That’s strange. In the manifest it is created a discovery match only on device name «PAX Calima». What devices are you discovering?

jimmyl88, 2023-02-08:
I am running HA in docker/container using HACS
Home Assistant 2023.2.3
Frontend-version: 20230202.0 - latest

Added this integration (https://github.com/eriknn/ha-pax_ble) Version "d213802" yesterday.
[image]

eriknn, 2023-02-08:
Okay - is each discovered device the same? Check the MAC-address. Did you originally add it/them manually?

eriknn, 2023-02-08:
I hadn’t upgraded my HA yet - did now and there seems to be some other issues as well. I will take a look at it later/tonight if I find the time!

jimmyl88, 2023-02-08:
Both of my Pax Calima devices have a MAC-address with starting "58:2B:DB:02:xx:xx", all the others discovered devices have different MAC-adresses. Thanks for helping out :)

eriknn, 2023-02-08:
I think debugging bluetooth might give a clue:
In configuration.yaml:
logger:
  default: warning
  logs:
    homeassistant.components.bluetooth: debug

eriknn, 2023-02-08:
I'm not able to reproduce this - I need to see your debug data to figure out what is happening!

eriknn, 2023-02-09:
Okay, I’m running HassOS with the Bluetooth integration. I don’t know how this compares and have you enable debugging on your system…

jimmyl88, 2023-02-10:
2023-02-10 07:28:41.825 DEBUG (MainThread) [homeassistant.components.bluetooth.scanner] Initializing bluetooth scanner with {'detection_callback': <bound method HaScanner._async_detection_callback of <homeassistant.components.bluetooth.scanner.HaScanner object at 0x7f2c07b30a60>>, 'scanning_mode': 'active', 'adapter': 'hci0'}
2023-02-10 07:28:41.826 DEBUG (MainThread) [homeassistant.components.bluetooth.scanner] hci0 (48:51:B7:7B:C1:F3): Starting bluetooth discovery attempt: (1/3)
2023-02-10 07:28:41.900 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] Registering scanner hci0 (48:51:B7:7B:C1:F3)
2023-02-10 07:28:41.971 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 3A:11:5A:8B:41:16 AdvertisementData(manufacturer_data={6: b'\x01\t!*\xf7\x9d\x0f\xe8\xd4\xe0W377706'}, rssi=-80) match: set()
2023-02-10 07:28:42.078 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: FC:2E:A9:3F:B7:33 AdvertisementData(local_name='PAX Calima', manufacturer_data={887: b'\x07\x00\x00\x053\xb7?\xa9.\xfc\x12\\\xf2\xa3O(\x00'}, service_uuids=['31ba0001-6085-4726-be45-040c957391b5'], rssi=-81) match: {'pax_ble'}
2023-02-10 07:28:42.078 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: DA:5F:1C:A6:0F:6F AdvertisementData(local_name='PAX Calima', manufacturer_data={887: b'\x07\x00\x00\x02o\x0f\xa6\x1c_\xda\x12Z\xcd\xfa\xa5\x0f\x00'}, service_uuids=['31ba0001-6085-4726-be45-040c957391b5'], rssi=-74) match: {'pax_ble'}
2023-02-10 07:28:42.145 DEBUG (MainThread) [homeassistant.components.bluetooth.wrappers] 58:2B:DB:02:1B:9C - PAX Calima -> /org/bluez/hci0: Connecting (last rssi: -127)
2023-02-10 07:28:42.148 DEBUG (MainThread) [homeassistant.components.bluetooth.wrappers] 58:2B:DB:02:16:FC - PAX Calima -> /org/bluez/hci0: Connecting (last rssi: -127)
2023-02-10 07:28:42.416 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 63:63:63:84:65:E3 AdvertisementData(manufacturer_data={224: b'\x01\xeb\xca\x9as\xe6'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-80) match: set()
2023-02-10 07:28:42.416 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 62:A2:A2:C0:FE:A4 AdvertisementData(manufacturer_data={224: b'\x01\xa6\xca\x81T\x91'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-66) match: set()
2023-02-10 07:28:42.416 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 4B:50:E9:50:C8:04 AdvertisementData(local_name='P mesh', manufacturer_data={76: b'\x10\x06u\x1e\x1e\xf3\xceb'}, service_uuids=['31ba0001-6085-4726-be45-040c957391b5'], tx_power=6, rssi=-77) match: set()
2023-02-10 07:28:42.510 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 44:5E:D3:E1:7C:F2 AdvertisementData(manufacturer_data={224: b'\x01\xdc\xca\x85dn'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x02VsTJ475FNXA\x00\x00\x01\x86:;&\x97'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-75) match: set()
2023-02-10 07:28:42.539 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 71:71:71:5E:87:C8 AdvertisementData(service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-88) match: set()
2023-02-10 07:28:42.539 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: EE:FC:9B:30:D9:73 AdvertisementData(local_name='PAX Calima', manufacturer_data={887: b'\x07\x00\x00\x01s\xd90\x9b\xfc\xee\x12V \xe1\x19c\x00'}, service_uuids=['31ba0001-6085-4726-be45-040c957391b5'], rssi=-78) match: {'pax_ble'}
2023-02-10 07:28:42.832 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 64:81:13:DE:EC:BC AdvertisementData(manufacturer_data={224: b'\x01\xcb\xca\x95\xa1\xde'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-68) match: set()
2023-02-10 07:28:42.832 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 67:A7:A7:54:83:FA AdvertisementData(manufacturer_data={224: b'\x01{\xca]\xf2G'}, service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-84) match: set()
2023-02-10 07:28:42.833 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 20:4E:F6:D7:87:6E AdvertisementData(local_name='PAX Calima', service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b3 [... cut ...] x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-78) match: set()
2023-02-10 07:28:44.332 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (48:51:B7:7B:C1:F3) [connectable]: 4C:CC:CC:C8:C2:F6 AdvertisementData(service_data={'0000fe9f-0000-1000-8000-00805f9b34fb': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000fe9f-0000-1000-8000-00805f9b34fb'], rssi=-91) match: set()

eriknn, 2023-02-10:
Okay, you are discovering lots og devices with name «PAX Calima» but which are obviously not yours. For instance MAC EE:FC:9B:30:D9:73.

Do you happen to live in a big new’ish apartment complex?

jimmyl88, 2023-02-10:
I live in a house with many devices, EE:FC:9B:30:D9:73 for example is one of many plejd-dimmers. 

Maby you only should look for MAC starting 58:2B:DB according to [https://www.cleancss.com/mac-lookup/58-2B-DB](url)

eriknn, 2023-02-10:
Strange, it would be nice to find out why it’s reporting like it does… But I can modify it, no problem!

eriknn, 2023-02-10:
Actually, it seems that it's not possible to filter discovery by mac. I could use manufaturer ID, but it doesn't advertise any (and I can not find an ID for PAX in online lists).

Most of the bluetooth integrations in HA use local_name, so I would like to keep it like that! Can you try to find out what is wrong on your side?

jimmyl88, 2023-02-14:
I have tried to disable discovery mode on my ubuntu server (running Bluez) as a workaround but it seems to be a bug in ubuntu as well (https://bugs.launchpad.net/ubuntu/+source/indicator-bluetooth/+bug/1787867).

It seems that all my devices is presenting without a name except my 3 Plejd-dimmers "P mesh".

I ran "btmon" and could se some difference between PAX and other devices as Apple, google

`
> HCI Event: LE Meta Event (0x3e) plen 38                                                                                                       #719 [hci0] 10.424209
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 58:2B:DB:02:1B:9C (Pax AB)
        Data length: 26
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Device Information (0x180a)
        Appearance: Unknown (0x0000)
        TX power: -2 dBm
        Name (complete): PAX Calima
        RSSI: -75 dBm (0xb5)

> HCI Event: LE Meta Event (0x3e) plen 36                                                                                                       #720 [hci0] 10.426205
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: F0:B3:EC:0C:67:9B (Apple, Inc.)
        Data length: 24
        Flags: 0x1a
          LE General Discoverable Mode
          Simultaneous LE and BR/EDR (Controller)
          Simultaneous LE and BR/EDR (Host)
        TX power: 4 dBm
        Company: Apple, Inc. (76)
          Type: Unknown (15)
          Data: c00a42fb9700040c
          Type: Unknown (16)
          Data: 0904
        RSSI: -68 dBm (0xbc)`

I have also tried to deactivate with no success.
[image]

Is it possible you could add a similar function like this for powercalc?
[image]

https://homeassistant-powercalc.readthedocs.io/en/latest/quick-start.html

eriknn, 2023-02-14:
I’ll look into it!

eriknn, 2023-02-14:
I don’t think function in the integration you are referrring to works the same way as what you want - it will disable a discovery function within the integration, not discovery of the integration itself.

What you want should be handled by the Bluetooth integration, but it currently has no such feature…

My guess is that there’s a bug in the BT integration, where if a device is not reporting 
complete/local name, the name of the previous device is used. If so then this is a bug that really should be reported and solved! This is your best way forward!

Your second option is to file a feature request for the BT integration or HA in general allowing autodiscovery of certain integrations to be disabled.

Your third option is to just select ignore on each discovered device, then they shouldn’t bother you anymore.

I really don’t see a good fix from my integration, sorry!

eriknn, 2023-02-14:
If you'd like to look at the process yourself, the integration match and subsequent discovery notification happens here:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/bluetooth/manager.py#L536

eriknn, 2023-02-22:
Closing this as the problem is most likely higher up in the HA-chain.

closed 2023-02-22 by eriknn