eriknn/ha-pax_ble #10: Bug: does not work after reboot

luotinen, 2023-02-22:
I was able to get one unit working, but after reboot the unit stops working. The unit works normally and is connectable using the Pax Calima mobile phone app.

Log snippet:

Logger: homeassistant.config_entries
Source: custom_components/pax_ble/__init__.py:31
Integration: Pax Bluetooth ([documentation](https://github.com/eriknn/ha-pax), [issues](https://github.com/eriknn/ha-pax/issues))
First occurred: 16.21.45 (1 occurrences)
Last logged: 16.21.45
Error setting up entry PAX Calima KH for pax_ble

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/__init__.py", line 29, in async_setup_entry
    await coordinator.read_deviceinfo()
  File "/config/custom_components/pax_ble/coordinator.py", line 109, in read_deviceinfo
    await self._fan.connect()
  File "/config/custom_components/pax_ble/calima.py", line 108, in connect
    ret = await self._dev.connect()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 249, in connect
    connected = await super().connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 206, in connect
    await self.get_services(
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 590, in get_services
    self.services = await manager.get_services(
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 573, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 663, in _wait_for_services_discovery
    done, pending = await asyncio.wait(
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 384, in wait
    return await _wait(fs, timeout, return_when, loop)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 491, in _wait
    await waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/__init__.py", line 28, in async_setup_entry
    async with async_timeout.timeout(10):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/pax_ble/__init__.py", line 31, in async_setup_entry
    _LOGGER.debug("Failed when loading initdata: " + err)
TypeError: can only concatenate str (not "TimeoutError") to str

eriknn, 2023-02-22:
This is only when loading device data - sw/fw version etc. This times out but should still work, but it seems I handled the timeout poorly so it crashes and everything dies.

I’ll fix this tonight. If you want, you can try to increase timeout in init.py:
async with async_timeout.timeout(10):

Change 10 to something higher (seconds)

eriknn, 2023-02-22:
Should be fixed now, please test!

luotinen, 2023-02-22:
Hmm... I have now succesfully added three units. After a couple of reboots, I have lost two of them. They report all sensors as "unknown". There are no error messages though. One of the three is still working normally. It is physically the closest. 

Logs:
2023-02-22 23:25:43.284 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 23') while processing template 'Template("{% set value = state_attr("sensor.shf_electricity_price_now", "today_prices")[ now().hour ] %} {{ value <= states("input_number.shf_price_slider") | float }}")' for attribute '_state' in entity 'binary_sensor.shf_price_acceptable'
2023-02-22 23:25:43.353 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: None has no element 23 when rendering '{{ state_attr("sensor.shf_electricity_price_now", "today_prices")[now().hour] }}'
2023-02-22 23:30:26.231 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:30:36.261 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:30:45.166 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!

luotinen, 2023-02-22:
After some waiting (waiting 10 minutes) the sensors reappear. The main log:

2023-02-22 23:25:43.284 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 23') while processing template 'Template("{% set value = state_attr("sensor.shf_electricity_price_now", "today_prices")[ now().hour ] %} {{ value <= states("input_number.shf_price_slider") | float }}")' for attribute '_state' in entity 'binary_sensor.shf_price_acceptable'
2023-02-22 23:25:43.353 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: None has no element 23 when rendering '{{ state_attr("sensor.shf_electricity_price_now", "today_prices")[now().hour] }}'
2023-02-22 23:30:26.231 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:30:36.261 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:30:45.166 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!
2023-02-22 23:35:44.358 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!
2023-02-22 23:35:58.986 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!
2023-02-22 23:35:59.262 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!
2023-02-22 23:41:02.132 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:41:11.114 ERROR (MainThread) [frontend.js.latest.202302020] :0:0 ResizeObserver loop completed with undelivered notifications.
2023-02-22 23:41:14.374 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Not connected!
2023-02-22 23:41:14.528 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching Device information: Characteristic with UUID 21 could not be found!
2023-02-22 23:41:19.711 ERROR (MainThread) [frontend.js.latest.202302020] :0:0 ResizeObserver loop completed with undelivered notifications.

Debug log:
https://www.dropbox.com/s/ik1hvad0zhcncay/home-assistant_pax_ble_2023-02-22T21-42-31.969Z.log?dl=0

eriknn, 2023-02-24:
Is this solved, or are there remaining issues? You might consider closing this one and create a new with current status, as many changes have been implemented lately.

luotinen, 2023-02-24:
Appears to be working now.

closed 2023-02-24 by luotinen