eriknn/ha-pax_ble #8: Bug: Intergration cannot be added

luotinen, 2023-02-22:
I installed the custom repository using HACS, then reboot. When I try to add the integration, I get this error:
{"message":"Invalid handler specified"}

eriknn, 2023-02-22:
I did a lot of changes yesterday, as there were some really bad code there, so I probably f’ed up something. Worked well when I tested though… Any chance you can extract some more debug logs?

luotinen, 2023-02-22:
This was a fresh new install.

Here is a log excerpt:

2023-02-22 10:49:52.002 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 10') while processing template 'Template("{% set value = state_attr("sensor.shf_electricity_price_now", "today_prices")[ now().hour ] %} {{ value <= states("input_number.shf_price_slider") | float }}")' for attribute '_state' in entity 'binary_sensor.shf_price_acceptable'
2023-02-22 10:49:52.045 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: None has no element 10 when rendering '{{ state_attr("sensor.shf_electricity_price_now", "today_prices")[now().hour] }}'
2023-02-22 10:49:52.704 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration pax_ble: No module named 'custom_components.pax_ble.calima'
2023-02-22 10:49:54.901 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration pax_ble: No module named 'custom_components.pax_ble.calima'
2023-02-22 10:49:54.956 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 925, in async_create_flow
    integration.get_platform("config_flow")
  File "/usr/src/homeassistant/homeassistant/loader.py", line 749, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 766, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/pax_ble/__init__.py", line 6, in <module>
    from .coordinator import PaxCalimaCoordinator
  File "/config/custom_components/pax_ble/coordinator.py", line 7, in <module>
    from .calima import Calima
ModuleNotFoundError: No module named 'custom_components.pax_ble.calima'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/discovery_flow.py", line 71, in _async_start
    await gather_with_concurrency(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 201, in gather_with_concurrency
    return await gather(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 199, in sem_task
    return await task
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 798, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 816, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 932, in async_create_flow
    raise data_entry_flow.UnknownHandler
homeassistant.data_entry_flow.UnknownHandler
2023-02-22 10:49:55.135 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration pax_ble: No module named 'custom_components.pax_ble.calima'
2023-02-22 11:17:00.923 ERROR (Thread-8) [pychromecast.socket_client] [Trio(192.168.68.103):32171] Error reading from socket.
2023-02-22 11:17:00.937 WARNING (Thread-8) [pychromecast.socket_client] [Trio(192.168.68.103):32171] Error communicating with socket, resetting connection
2023-02-22 11:17:01.052 ERROR (Thread-8) [pychromecast.socket_client] [Trio(192.168.68.103):32171] Failed to connect to service ServiceInfo(type='host', data=('192.168.68.103', 32171)), retrying in 5.0s
2023-02-22 12:02:43.701 ERROR (Thread-5) [pychromecast.socket_client] [DNA TV-hubi(192.168.68.115):8009] Error reading from socket.
2023-02-22 12:02:43.707 WARNING (Thread-5) [pychromecast.socket_client] [DNA TV-hubi(192.168.68.115):8009] Error communicating with socket, resetting connection
2023-02-22 12:02:43.720 ERROR (Thread-5) [pychromecast.socket_client] [DNA TV-hubi(192.168.68.115):8009] Failed to connect to service ServiceInfo(type='host', data=('192.168.68.115', 8009)), retrying in 5.0s
2023-02-22 12:07:21.979 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration pax_ble: No module named 'custom_components.pax_ble.calima'

eriknn, 2023-02-22:
Ah, I see the error. The file Calima.py should have been renamed to calima.py (all lowercase). For some reason this didn’t get pushed! Will fix ASAP!

closed 2023-02-22 by luotinen