eriknn/ha-pax_ble #13: Feature request: update sensors after user changes parameters

luotinen, 2023-02-23:
It would be great to ask for an additional sensor reading update, for example, 20-30 seconds after the user has changed any parameter values.

Currently:
1. User activates Boost mode
2. The boost mode indicator is turned on
3. RPM is unchanged
4. Several minutes elapse with incorrect RPM reading
5. The RPM reading and mode are updated when the next polling time is met

Proposed:
1. User activates Boost mode (or makes any other change in the parameters)
2. Incorrect mode and RPM are shown until the next polling
3. Additional polling is done with a small delay
4. The RPM reading and mode are updated when the next polling time is met

In addition, sometimes the Boost mode fails. Currently it goes like this:
1. User activates Boost mode
2. If this fails, the fan does not start spinning faster but the boost mode indicator is still turned on
3. For several minutes the Boost mode is displayed on despite the fan not actually spinning faster
4. The Boost mode and Device mode are updated to correct values when the next polling time is met

Proposed:
1. User activates Boost mode
2. Check if the fan is in boost mode, if yes, keep the indicator to Boost mode, otherwise drop it back down to not in boost mode

eriknn, 2023-02-23:
As og latest version, all commands should revert if they fail!

Will look into an extra poll after a command!

luotinen, 2023-02-23:
Some kind of middle ground could be to keep the connection for a while while the user is interacting with the device and then return to intermittent polling later. I believe most users will want to retain the possibility of using the mobile phone app.

luotinen, 2023-02-24:
I can confirm the settings revert back if the calls are unsuccessful. There is a small delay, but it is not too annoying. I wonder how this would go together with automation and scripting: at 22:00 set the RPM to some value -> fail -> then what? Probably the automation should just be set: after 22:00 every 5 minutes if RPM is not at desired value, attempt to set it (?)

I put all three to boost mode. It took 5 min for 1st fan to report the correct RPM, 10 min for 2nd fan, and 3rd fan never reported the correct RPM reading (Boost mode auto-expired). I also checked with the mobile phone app and the new RPM readings were available over Bluetooth shortly after activating the Boost mode.

Also, even after the Boost mode expiring, the 3rd fan still claims the Boost mode is on. Interestingly the mode = Trickle ventilation.

luotinen, 2023-02-24:
Finally, after 31 minutes, the 3rd fan reports the correct Boost mode, RPM, and Mode! It I can confirm that eventually the fans will report the correct state. It might be tricky to use these for automation as the fans might not respond to commands or report the modes, RPMs, etc.

eriknn, 2023-02-25:
Can you try the latest version? I made some significant changes that keep the connectiong for ~100 sec after a write, and polls every 10 second. A read takes some time, so I don’t think I can lower it much more. HA has a minimum of 5 sec anyways.

This changes should improve some other connection issues as well :)

luotinen, 2023-02-25:
I did a clean reinstall of pax_ble. Even after multiple reboots I am not able to add even one device. For all three sensors I get "Connection failed". The logs are clean so it does not seem to trigger anything.

How can I revert back to the previous version?

luotinen, 2023-02-25:
This started working unexpectedly, and I was able to test this now. It works like this:
1. On the first command, if the command goes through, there is a delay of about 30...60 seconds before the sensors are updated. It is long, but quite acceptable and much, much better than in the previous version.
2. On subsequent commands shortly after, the device is very responsive. I counted it takes about 8-10 seconds only for the sensors to update after the command is confirmed.

=> This is now implemented. Thanks again!

closed 2023-02-25 by luotinen

eriknn, 2023-02-25:
I don’t when things work «unexpectedly». Don’t like your delays either!

I suspect it is because you have so many fans, at least that’s one difference between our integrations.

On my setup I have the normal poll rate to 300s (default). If I enable boostmode, I will get a new reading with updated sensor data after 10s, exactly as expected.

I also see that regular polls (with connect/disconnect) take anywhere from 3.5 -  5.5 seconds. The «quick polls», where we already are connected, and we don’t disconnect take roughly 1.5 seconds.
What are your numbers on this?

luotinen, 2023-02-26:
Poor choice of words. Unexpectedly => Due to reasons not related to this ticket.

Too many fans theory: I unplugged two fans, leaving only one fan plugged. Then it takes 15 seconds for the initial command to enable boost mode and 11 seconds to disable it. To re-enable, 11 seconds. To re-disable 11 seconds.

Then I replugged all three the fans, waited 100 seconds, and:
Fan #1: 17 sec, 12 sec, 12 sec, 11 sec
Fan #2: 16 sec, 10 sec, 11 sec, 11 sec
Fan #3: 14 sec, 10 sec, 11 sec, 11 sec

=> Seems to work normally now. I did not make any changes since yesterday.

PS. My hardware is Raspberry Pi 3B. The distances to the fans are 4 m, 6 m and 9 m. Only lightweight walls in between (no rebar or metallic major obstructions). The distance to the 4th fan is 12 m and there are two brick walls in between.

eriknn, 2023-02-26:
Great to hear! I believe that the command is much quicker than what you said - it’s just that we wait a bit long before we try to read the new data. 

I’m considering something like this:
- Command sent
- Read data after few seconds, maybe between 2-5 sec
- 10 sec between next 9 reads or so
- Back to normal poll rate

I’m not sure if very frequent reads could influence other bluetooth integrations, that’s why I’m a bit careful with the delays.