eriknn/ha-pax_ble #14: Feature request: human readable units

luotinen, 2023-02-23:
For the users, could you provide also l/s and m3/h sensor readings converted from the fan RPM readings. Based from the mobile app the conversions are:
RPM	l/s	m3/h
991	10	36
1037	11	40
1122	12	43
1217	13	47
1334	14	50
1404	15	54
1446	16	58
1499	17	61
1569	18	65
1662	19	68
1755	20	72
2400	30	108

luotinen, 2023-02-23:
Here is the rest of the data:
RPM	l/s	m3/h
991	10	36
1037	11	40
1122	12	43
1217	13	47
1334	14	50
1404	15	54
1446	16	58
1499	17	61
1569	18	65
1662	19	68
1755	20	72
1789	21	76
1856	22	79
1931	23	83
1999	24	86
2100	25	90
2144	26	94
2209	27	97
2269	28	101
2327	29	104
2400	30	108

Looks like:
l/s = 0.0141 x RPM + 4.2511 (R^2 = 0.9972)
m3/h = 3.6 x l/s

l/s should be displayed as full integers
m3/h should be displayed as full integers

luotinen, 2023-02-23:
Alternatively using only the minimum setting 975 rpm (10 l/s) and maximum setting 2400 rpm (30 l/s) only, the formulas are:
l/s = 0.014 x RPM - 3.642
m3/h = 3.6 x l/s

eriknn, 2023-02-23:
Those values refer to a free flowing fan with no obstructions. Real life values are lower and impossible to guess. I think this could be misleading….

luotinen, 2023-02-23:
I have a flowmeter (Trotec BA06) and the readings are quite close. For example right now:
WC1 fan, setting 16 l/s, measurement 17,7 l/s
WC2 fan, setting 10 l/s, measurement 7,6 l/s
Bathroom fan, setting 16 l/s, measurement 18,6 l/s

Those values are available in the app and they are needed to adjust the ventilation. Most common unit (in Finland) is l/s. For example about 6 l/s /person will keep the CO2 levels within acceptable limits.

Could the calculated sensor say:
"Freeflow speed: XXX l/s (xxx m3/h)"

Also, another built-in assumption is that everyone has 100 mm ducts. If the duct is something else, then the flow will also differ. For me, the WC1 and bathroom fans have a rectangle profile larger than 100 mm circular duct. That could explain the difference between the values.

eriknn, 2023-02-23:
Sure, I’ll take a look at it, shouldn’t take more than a minute to implement. Might have time tonight!

eriknn, 2023-02-23:
Added now. Ended up with m^3/h as this is the SI unit and what is defined in the HASS consts.

closed 2023-02-24 by luotinen