eriknn/ha-pax_ble #17: Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble

roosfeld, 2023-02-28:
After latest update I got error message and integration is not working. Could you please check?

From logs:

Error setting up entry PAX Calima for pax_ble
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/pax_ble/__init__.py", line 30, in async_setup_entry
    scan_interval_fast = config_entry.data[CONF_SCAN_INTERVAL_FAST]
KeyError: 'scan_interval_fast'

[image]
[image]

eriknn, 2023-02-28:
Hi, I thought my update comment was visible?
I made changes to the config flow, added a variable. This means you must probably remove and add the fan again, so that the new variable is added to the config.
I think maybe there are ways I vould have done this without breaking it, but I’m haven’t read up on those parts of integrations :/

roosfeld, 2023-02-28:
Thanks for the quick response and I managed to add the devices again, everything works again. Its a good feature and I'm very happy with the work that you have done. Thanks! If you need help with testing then I happy to help.

eriknn, 2023-02-28:
Great, I’ll close the issue then 😀

closed 2023-02-28 by eriknn