eriknn/ha-pax_ble #18: KeyError: 'scan_interval_fast'

lazi3b0y, 2023-03-04:
Hey, updated to the latest version a few days ago and noticed this in the logs today along with the fans not being setup in HA

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/pax_ble/__init__.py", line 30, in async_setup_entry
    scan_interval_fast = config_entry.data[CONF_SCAN_INTERVAL_FAST]
KeyError: 'scan_interval_fast'

Any ideas? :)

eriknn, 2023-03-04:
https://github.com/eriknn/ha-pax_ble/issues/17

lazi3b0y, 2023-03-05:
> #17
> 
> I’ll try to use automatic version control in config flow the next time - that should help avoid breaking stuff like this.

Ah, cool, it work after removing them and adding them again, as you said :) Thx! 
No worries, you're quick to respond/fix which I think is more important than not breaking things!

closed 2023-03-05 by lazi3b0y