eriknn/ha-pax_ble #19: connection issues with bluetooth proxies

gribber, 2023-03-10:
Got some connection issues, not yet sure if its related to the esphome bluetooth proxy or the integration
currently using 3 proxies, sometimes it works just fine, next hour it might just spam "Error when fetching data: Not connected" all the time, I would say it work about 20% of the time right now

I'm not sure how home assistant handling connections if the same device is seen from different proxies, might be one of the issues (or if this integration can handle that?)

The other error I get is "name 'err' is not defined", which seems to be related to the integration, at least that's the one who outputs the error :-)

debug logs below: 

```
2023-03-10 09:59:43.738 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching data: Not connected
2023-03-10 11:23:31.450 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
bleak.exc.BleakError: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish
2023-03-10 11:23:31.451 ERROR (MainThread) [custom_components.pax_ble.coordinator] Unexpected error fetching Pax Calima data: name 'err' is not defined
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 546, in bluetooth_device_connect
    await event.wait()
  File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 545, in bluetooth_device_connect
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/coordinator.py", line 64, in _async_update_data
    await self.async_fetch_data(disconnect=not self._fast_poll_enabled)
  File "/config/custom_components/pax_ble/coordinator.py", line 268, in async_fetch_data
    if not await self._fan.connect():
  File "/config/custom_components/pax_ble/calima.py", line 112, in connect
    ret = await self._dev.connect()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 249, in connect
    connected = await super().connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 100, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 314, in connect
    await self._client.bluetooth_device_connect(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 556, in bluetooth_device_connect
    await event.wait()
  File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/coordinator.py", line 63, in _async_update_data
    async with async_timeout.timeout(30):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 239, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/pax_ble/coordinator.py", line 66, in _async_update_data
    _LOGGER.debug("Failed when fetching sensordata: " + str(err))
NameError: name 'err' is not defined
2023-03-10 11:54:44.451 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
bleak.exc.BleakError: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish
2023-03-10 11:54:44.452 ERROR (MainThread) [custom_components.pax_ble.coordinator] Unexpected error fetching Pax Calima data: name 'err' is not defined
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 546, in bluetooth_device_connect
    await event.wait()
  File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 545, in bluetooth_device_connect
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/coordinator.py", line 64, in _async_update_data
    await self.async_fetch_data(disconnect=not self._fast_poll_enabled)
  File "/config/custom_components/pax_ble/coordinator.py", line 268, in async_fetch_data
    if not await self._fan.connect():
  File "/config/custom_components/pax_ble/calima.py", line 112, in connect
    ret = await self._dev.connect()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 249, in connect
    connected = await super().connect(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 100, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 314, in connect
    await self._client.bluetooth_device_connect(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 556, in bluetooth_device_connect
    await event.wait()
  File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/pax_ble/coordinator.py", line 63, in _async_update_data
    async with async_timeout.timeout(30):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 239, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/pax_ble/coordinator.py", line 66, in _async_update_data
    _LOGGER.debug("Failed when fetching sensordata: " + str(err))
NameError: name 'err' is not defined
2023-03-10 12:25:17.449 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching data: Not connected!
2023-03-10 13:21:43.449 WARNING (MainThread) [custom_components.pax_ble.coordinator] Error when fetching data: Not connected!
```

gribber, 2023-03-10:
fixed the exception, missing definition, #20 
I'll see if that clear things out

eriknn, 2023-03-10:
Yeah that was a bad mistake on my behalf - too much copy-paste I guess.
Won't do much to fix your issues though :/

helmuto-maker, 2023-03-11:
I also try to access Pax Calima through Bluetooth proxy. Worked fine in the beginning but now there is no connection. I can see that the ESPhome proxy is discovering the Calima fan. But I get this error:

Logger: homeassistant.config_entries
Source: custom_components/pax_ble/__init__.py:32
Integration: Pax Bluetooth ([documentation](https://github.com/eriknn/ha-pax), [issues](https://github.com/eriknn/ha-pax/issues))
First occurred: 16:35:23 (1 occurrences)
Last logged: 16:35:23
Error setting up entry PAX Calima for pax_ble

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/pax_ble/__init__.py", line 32, in async_setup_entry
    scan_interval_fast = entry.data[CONF_SCAN_INTERVAL_FAST]
KeyError: 'scan_interval_fast'

eriknn, 2023-03-11:
I added a config parameter that is missing in your hass configuration. You need to remove the fan and add it again to create this parameter.

closed 2023-03-11 by eriknn

helmuto-maker, 2023-03-12:
Thanks  - Erik. It works now !