eriknn/ha-pax_ble #22: Latest release

eriknn, 2023-03-22:
The latest release has a total rework of the configuration flow - that is, how the integration and devices are added and stored in the system.
I had an attempt on the same for a long time ago, but it was too much work for my need, but as I noticed more people are starting to use the integration, I would like for everything to be a bit smoother - for instance I recently added icons for the integration.

The main purpose of the latest change is to store all devices inside one configuration flow. This means you will now only have one instance of the integration, but several devices. Not sure how much this matters to you, but for me this feels alot more "correct".

Note that there were a lot of changes made to make this happen, and even though I tested as good as I could with my one fan, I'm sure you guys will find som bugs, so please bear with me and just let me know what happens.

I have found one minor issue myself - whenever you add/remove/edit a device you should wait at least 1 minute before remotting, or the changes aren't saved to disk. This makes the configuration out of sync, and you might have to remove it and start over again to fix it, depending on what changes were made.
See the following:
https://github.com/home-assistant/core/issues/90078

moorgrove, 2023-04-05:
First thanks for a great plugin!

For your knowledge. Even if I wait more than 1 minute after adding the fan it will not be saved. It works as it should unit I reboot. After that it stays in the Integration panel but loses connection and a new one device is auto detected. I can't see anything in the logs.

eriknn, 2023-04-05:
@moorgrove 
Okay, do you have only one device? The storage issue should only happen if you added the integration before you added the fan.

Try removing the integration and restarting. The device should then be detected. Add it, and it’ll add the integration with one device. This should work well.

eriknn, 2023-04-12:
I believe I fixed this issue now!

closed 2023-09-04 by eriknn