eriknn/ha-pax_ble #23: Boostmode can not be enabled if boostmode time is set higher than 900s.

Turizmo, 2023-04-22:
I tried setting boostmode time to 1800s, and then boostmode can not be enabled. I did some testing and 900s seems to be the highest i can go where boostmode works. If i go any higher, the fan tries to start, but stops before it even gets up to speed. The attribute for max boostmode time says 3600. It should be limited to 900 if that is the true max value, or is there another bug that causes this limitation?

eriknn, 2023-04-22:
Ah, good observation. The max of 3600 was rather arbitrarily chosen. It might be that 900 is max that the fan actually accept. I’ll change the max input when I find time for doing a minor change!

eriknn, 2023-09-04:
This was fixed shortly after the previous posts.

closed 2023-09-04 by eriknn