eriknn/ha-pax_ble #25: Can´t add devices

zlakes01, 2023-07-27:
[image]

eriknn, 2023-07-27:
Was it discovered automatically? If not, check the bluetooth logs to see discovered devices. I’m guessing it’s out of range…

zlakes01, 2023-07-27:
Out of range is possible. I have to check it.

zlakes01, 2023-07-28:
I couldn´t find bluetooths logs. Where is it?

ivoh95, 2023-07-28:
I have a similar issue. It turns out I  had somehow gotten the fan in a buggy state where neither the phone app would find it nor HA. Power cycling the fan fixed it and it was automatically discovered then. Something easy to try if the phone app works at that distance or if a power cycle helps

eriknn, 2023-07-29:
@zlakes01 Under "Bluetooth" and your adapter - select "Download diagnostics" or similar. This should show all discovered devices and signal strength (RSSI). Not sure how reliable that value is though.

@ivoh95 I have experienced the same thing once. But I've been doing lots of connecting/reconnecting while testing, so I'm not surprised that it was too much for the fan :)

zlakes01, 2023-07-31:
It works after an update of Core. I see all three Pax Calima and can manage them.

closed 2023-09-04 by eriknn

catellie, 2023-12-11:
May I chime in with a question here? I've had this integration working just fine for a long time (6+ months) and then I had to move my server a couple of meters sideways (yeah, it wasn't really pretty where I had it originally). Technically, it's actually slightly closer than before, though there is a bit more metal in the "line of sight" (two floors up partly through cement floor).
Interestingly, my io9 toothbrush is actually somewhat closer, but never had any issue. Anyway, after the move it went very glitchy so I tried to remove it in order to start over - bad idea I guess because now it pops up as a new device every day or so, but I never managed to reconnect it again. For kicks I tried installing a ESP32 M5Stack Atom Lite BT proxy which appears to make all the right moves, but still no connection. Do you @eriknn know whether it should be able to handle the Calima? Their app works fine and I can't _really_ say I need this integration much, I'm just sad it worked fine before the move and now it fails utterly.

eriknn, 2023-12-12:
Sorry but I don’t know much about BT extenders, so I can really help you.
If the connection is that bad I would recommend using the ignore-function on the discovery, that should at least make it leave you alone :)