eriknn/ha-pax_ble #26: Possible bug with HA 2023.11 - sensor.abc_mode unavailable

BoneheadFraggle, 2023-11-04:
After updating to HA 2023.11, the sensor "mode" is unavailable. 

Possibly this part in log may be relevant? It only occurs once in the log, even though I have four Pax's (pax_badrum is one of them) and all four mode sensors are unavailable.

All other entities seem to work fine.
Restarts of Bluetooth, HA does not make any difference.

```
2023-11-04 12:58:03.496 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform pax_ble
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.pax_badrum_mode cannot be added as the entity category is set to config
2023-11-04 12:58:03.501 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up pax_ble platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 367, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.pax_badrum_mode cannot be added as the entity category is set to config
```

eriknn, 2023-11-04:
Yeah, looks like there’s been a change in one of the latest updates - will have a look at it at first opportunity!

eriknn, 2023-11-04:
Fixed now. HASS no longer accepts entities of type "CONFIG" that are not writable, i.e. sensors. I had to change the type of this one to "DIAGNOSTIC". Also in this update is a change where all "time"-values are changed from "Text"-entity to "Time"-entity. The time entity was added a couple of months ago I think, so I had that modification already done on my local version.

closed 2023-11-05 by eriknn