eriknn/ha-pax_ble #27: Vent Axia - Failed fetching sensordata.

DieterKoblenz, 2023-11-18:
```
2023-11-18 16:34:10.534 DEBUG (MainThread) [custom_components.pax_ble.calima] Connecting to 4C:96:2D:93:5D:6A
2023-11-18 16:34:11.290 DEBUG (MainThread) [custom_components.pax_ble.calima] Connected to 4C:96:2D:93:5D:6A
2023-11-18 16:34:11.290 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Couldn't read manufacturer! Characteristic with UUID 21 could not be found!
2023-11-18 16:34:11.723 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Svensa
2023-11-18 16:34:11.723 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading config data
2023-11-18 16:34:11.723 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when loading config data: Characteristic with UUID 90cabcd1-bcda-4167-85d8-16dcd8ab6a6b could not be found!
2023-11-18 16:34:11.723 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading sensor data
2023-11-18 16:34:11.803 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when fetching sensordata: unpack requires a buffer of 12 bytes
2023-11-18 16:34:11.804 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Pax Calima: Svensa data in 1.270 seconds (success: True)
```

I've just added this device, unfortunately. I am not getting any values. Reportedly, this should work.

eriknn, 2023-11-18:
Yes it seems that you are able to connect but not read the values. Are you getting any other values? The firmware/hardware versions could be interesting to know.

DieterKoblenz, 2023-11-18:
I've enabled debug logging, but this is the only log I received (ofcourse repeated over time)

It's the Vent-Axia Svara. 

This topic pointed me to this repo: https://community.home-assistant.io/t/vent-axia-svara-app-controlled-extractor-fan/260082

eriknn, 2023-11-18:
Ah, yes it’s clearly the same fan, bit they might have modifies the software a bit. If you are somewhat proficient in Python we could do some testing - with bluetooth we can ask the device for its UUID’s. Those UUIDs are what we need to make the request. Or maybe they have their own app? We could decompile that to find what we need.

ceedebee70, 2023-12-24:
Hi, same problem overhere, same fan, same log entries. Any solution yet?

eriknn, 2023-12-24:
No, need someone to test as I suggested above, then we could see what has changed.

ceedebee70, 2023-12-24:
I am not familiar with python but addpted your script a bit with you suggestion, this is the result:
2023-12-24 16:19:58.696 DEBUG (MainThread) [custom_components.pax_ble.calima] Result: b'\x00\x10B\x04\x03\x08\x00\x00\x00\x03\xee\x07\x00\x16\x00'
2023-12-24 16:19:58.698 DEBUG (MainThread) [custom_components.pax_ble.calima] Length: 15
2023-12-24 16:19:58.698 DEBUG (MainThread) [custom_components.pax_ble.calima] Unpacked: (0, 16, 66, 4, 3, 8, 0, 0, 0, 3, 238, 7, 0, 22, 0)
2023-12-24 16:19:58.876 DEBUG (MainThread) [custom_components.pax_ble.calima] Read Fan States: (4096, 1089, 2051, 0, 0, 60931, 7, 0, 22, 0)

But sadly no results in the entities yet.

ceedebee70, 2023-12-24:
Some more errors in the log:

 Couldn't read firmware revision! Characteristic with UUID 17 could not be found!
 Couldn't read hardware revision! Characteristic with UUID 15 could not be found!
 Couldn't read software revision! Characteristic with UUID 19 could not be found!

 Failed when loading config data: Characteristic with UUID 90cabcd1-bcda-4167-85d8-16dcd8ab6a6b could not be found!
 Failed when fetching sensordata: Characteristic with UUID 118c949c-28c8-4139-b0b3-36657fd055a9 could not be found!

eriknn, 2023-12-25:
If your test was with "CHARACTERISTIC_SENSOR_DATA", then your result is very different than what is expected.
You are receiving 15 bytes vs expected 12. Also, I don't see any bytes with values that looks like the temperature, at least not with the same scaling as my device.
I wonder if the data could be different if a different "mode" is selected on the fan, but I wouldn't think so.
It would be very nice to read out all the data I mentioned in my examle (model number, manufacturer etc). Also if you could try to figure out how your data is arranged and how it is scaled, that would help.

ceedebee70, 2023-12-30:
This was with "CHARACTERISTIC_SENSOR_DATA", tried to get the other data but no luck sofar.

eriknn, 2024-02-14:
I'm closing this as there's not much I can do about this. If someone is able to do some bluetooth testing to find out how this fan works, we could maybe consider adding some kind of fix / new device type.

closed 2024-02-14 by eriknn