eriknn/ha-pax_ble #29: Unable to add Svara

MedicoreV8, 2023-12-12:
I've installed the integration through HACS and restarted HASS.
The Vent-Axia Svara Fan works through the app, it's brand new.
Somehow the custom integration won't recognize the fan automatically and I have no idea how to add it by hand.

I'm Running:
HA Core version 2023.12.1
ha-pax_ble version a7aad78
My Bluetooth adapter works, it's an integrated chip on a NUC, I've had other devices connected through Bluetooth.

Where do I even add the device by hand?
If I open the HACS integration and click on Pax Bluetooth, then there is no option to add a device by hand.

MedicoreV8, 2023-12-12:
Before anyone tells me te factory default it because it is already paired with the app.
I tried adding it in Home assistant first, when that failed I added it via the app to make sure the device was in fact working.

MedicoreV8, 2023-12-12:
Okay, I've solved it.
After adding the custom integration through HACS you have to add the integration manually
Then you can manually add the fan as described in some other issues.

Could the documentation be a little more elaborate for those that don't work with HACS a lot?

closed 2023-12-12 by MedicoreV8

eriknn, 2023-12-12:
HACS just installs the integration. At this point I think auto-discovery of the device should be possible, and will also add the integration if completed. 
If you wish to add the first fan manually you must do as you said, and add the integration manually.