eriknn/ha-pax_ble #32: connect with Shelly Proxy

Guenni9, 2023-12-16:
Dear @eriknn,

As I can read here:

https://github.com/eriknn/ha-pax_ble/issues/21

@luotinen  managed to use the fan with a Shelly Gen2 Bluetooth-proxy?
May I ask you or him, how you did this? My fan works fine for test direct connected with the  RPi built in Bluetooth.
But I had to add it manually with MAC and key. It wasn't recognised automatically..

The Shelly doesn't work with the fan, but I have also two BLE door-sensors, that works fine with it. Therefore I think, the Proxy works.
I've also deleted the fan but trying to reinstall manually it shows an error. It's not possible to add it by hand. I think it is recognised, as I can see in Shelly log:
`{
            "name": "Vent-Axia Svara",
            "address": "58:2B:DB:35:A5:01",
            "rssi": -57,
            "advertisement_data": [
              "Vent-Axia Svara",
              {},
              {},
              [
                "0000180a-0000-1000-8000-00805f9b34fb"
              ],
              -2,
              -57,
              []
            ],
            "details": {
              "source": "34:98:7A:68:60:E8"
            }
          },
`
Could someone help me? Would be great!

Feel free to ask for more logs or Information.

Have a nice day!

eriknn, 2023-12-16:
Sorry but I have no knowledge on this topic.  I guess you should try @luotinen

Guenni9, 2023-12-16:
Okay, thank you! I hope @luotinen reads this an could help me :-)

But basically, how can I solve this error:
[image]
What could be the reason?

eriknn, 2023-12-16:
All we know is that the connection couldn’t be made. Usually this means signal out of range, but could probably be some other reasons as well.

el97, 2023-12-16:
Have you made sure that the shelly device is configured to act as a proxy? You need to set the scanning mode to either active or passive. read more here https://www.home-assistant.io/integrations/shelly#bluetooth-support

Guenni9, 2023-12-16:
Yes, it works well with two Xiaomi BLE door-sensors right now.
There are two door-sensors and one PAX fan in my garden house very nearby the Shelly Pro3EM.
The PAX also works fine, but only with the Android app, not via Shelly. But works also fine if direct coupled with the RPi4.
Sure, the proxy works fine..

MarkP12345, 2024-01-02:
Hi Guennu9, were you able to fix this issue? I have the same issue as you.

Guenni9, 2024-01-02:
Dear @MarkP12345 , 
as far as I understand this, Shelly BLE proxy only was able to read messages from devices and send it to home assistant. I doesn't support active mode. So, I was't able to use it, Maybe sometimes it will be improved:
[https://www.shelly-support.eu/forum/thread/18876-bluetooth-gateway-active-connection/?postID=195653#post195653](url)

eriknn, 2024-01-02:
Sorry to hear this - but does this mean that we can close this issue?

Guenni9, 2024-01-03:
@eriknn  I think for me it could be closed, thank you!

closed 2024-01-29 by Guenni9