eriknn/ha-pax_ble #34: I can't add my vent axia svara fan

MarkP12345, 2024-01-02:
Hi when I try to add my fan I get the message: failed to connect (see picture, I deleted the mac and pincode after the failed to connect message). Is there something I can do about this? thanks!
[image]

Hedda, 2024-01-06:
> I can't add my vent axia svara fan

Does that "Vent-Axia Svara" fan use the same board and firmware internally as the "Pax Calima"?

I see that both Pax Electro Products AB and Vent-Axia are both part of Volution Group PLC,( and so is a few other doughter companies too), so it would make sense that they use the same hardware, but if so do they also use the exact same firmware?

https://www.vent-axia.com/range/lo-carbon-svara

https://pax.se/sv/produkt/calima/pax-calima-flakt

https://community.home-assistant.io/t/pax-calima/118489/

https://community.home-assistant.io/t/vent-axia-svara-app-controlled-extractor-fan/260082

Prileyman, 2024-02-05:
Any luck in getting this fan to connect? I have the same issue

MarkP12345, 2024-02-06:
I replaced my bleutooth dongle and used an extensioncable and that did the
trick.

Prileyman, 2024-02-06:
Now working. The MAC address on the label was incorrect. Once using a Bluetooth app to get the correct address, ensuring no other program was running and entering the address correctly (all lower case with colons) it connected! Phew

Hedda, 2024-02-12:
> Now working. **The MAC address on the label was incorrect.** Once using a Bluetooth app to get the correct address, ensuring no other program was running and entering the address correctly (all lower case with colons) it connected!

@Prileyman recommend that you open a new issue for that with suggested possible solutions (like maybe converting/translating or force formatting of the MAC address) since your problem and solution might not be the same as the original poster.

eriknn, 2024-02-14:
Added hint on mac-adress format, also improved format handling.

closed 2024-02-14 by eriknn