eriknn/ha-pax_ble #35: Added thru HACS but no config screen

GeoffMorg, 2024-01-04:
I've successfully added the integration and he files are in custom components but when I click on the integration to configure it I get this screen.

[image]

eriknn, 2024-01-05:
I have no idea how you got to there - sure you disn’t press the wrong button somewhere?

Cptkex82, 2024-01-06:
When you manually add it to hacs it doesnt show any text as per the screenshot. 

But that only for installation, the configuration is when you add the integration in devices and integrations in configuration. Install in hacs, restart and it should show up automatically in integrations ready to config

GeoffMorg, 2024-01-06:
Ah OK, now I've got it. HACS is a very grey area for me!!

I was hoping it would find the device but no. So now I just need to pull the fan from the ceiling to get the MAC.

Thanks both

Cptkex82, 2024-01-06:
My was automatically detected but you still need the pin code from inside.

Quepali, 2024-02-10:
I also struggle finding any config. Can't find anything ha-pax or pax related anywhere. Cloned the code with terminal to custom_components\pax_ble and rebooted, but still nothing.

eriknn, 2024-02-10:
«Add integration» -> «Pax Bluetooth»

Quepali, 2024-02-10:
I got this searching now for the full name. Is this correct?
[image]

eriknn, 2024-02-10:
No. The integration seems to not be detected. Please verify that you put it in the correct location.

Quepali, 2024-02-10:
This path looks a bit strange...
[image]

Should all the files be under config/custom_components/pax_ble ? (as you can see I somehow got an extra custom_components/pax_ble in the path)

eriknn, 2024-02-10:
You’re correct, that’s your issue.

Quepali, 2024-02-11:
I suggest either not have the structure in the repo or adding a line to the documentation for when cloning. I don't know what the best solution is.

eriknn, 2024-02-14:
Removing this structure would put the integration-files in the same folder as the github-readme etc, so that won't work. Seems that most developers do it the same as me,
Also feel that what is in the documentation already would be good enough for most people, so I'll close this with no changes.

closed 2024-02-14 by eriknn