eriknn/ha-pax_ble #36: Fan stops responding

Kaaremi, 2024-01-30:
Had anybody else ran into a situation that Pax Calima stops communicating with Home Assistant and the Pax Calima android app via Bluetooth? 

I've had it twice during the last few weeks. The resolution is simple,, disconnecting the power supply for a moment. I wonder what might cause the issue.

eriknn, 2024-01-31:
I had the same during initial integration testing, caused by lots of connects and reconnects. This shouldn’t be an issue with the released version though. Are you writing often to any entities?

eriknn, 2024-01-31:
Okay, this is pretty much how I run my fan as well, with no issues. Maybe it’s some kind of bug in your fan FW…? Can you check which version it is reporting, just so we know for future reference?

Kaaremi, 2024-01-31:
Hi,

The Pax Wireless version is 3.2.0, but how can I check the fan's FW
version? Can't see it in the app.

BTW, when I tapped the "i" icon in the Android app's too left corner I got
the attached error message.

Parhain terveisin / Best Regards / Mit freundlichen Grüßen

Mikko Kaarela ***@***.***)

eriknn, 2024-01-31:
In Home Assistant, under the device, you’ll se FW and HW versions.
I don’t see any attached error messages.

Kaaremi, 2024-01-31:
Weird, that's where I tried to find the versions, but can't find them.

Parhain terveisin / Best Regards / Mit freundlichen Grüßen

Mikko Kaarela ***@***.***)

eriknn, 2024-01-31:
I have no idea, I don’t know anything about BT proxies.

eriknn, 2024-02-14:
I'm closing this as I don't think this is an issue that can be solved by the integration.

closed 2024-02-14 by eriknn