eriknn/ha-pax_ble #38: Releases

JanPetterMG, 2024-02-14:
Hi,
Is it possible for you to start publishing tagged releases? I'm receiving updates in Home Assistant for each code commit, because there's no better alternative offered at the moment, but this also means there's an high risk of bugs, and it's challenging to see what's changed or if the committed code is work in progress or even tested.

Please tag and create a new releases here: https://github.com/eriknn/ha-pax_ble/releases

eriknn, 2024-02-14:
Nah, shouldn’t be necessary. All releases should be

JanPetterMG, 2024-02-15:
Please read it one more time, I'm asking for you to create tags so that we DON'T need to download your WIP code.

eriknn, 2024-02-15:
My work in progress is stored locally on my computer, so I'd be impressed if you downloaded it. Up until now, any time I have made a commit it would mean a new release. I'll check out the release function next time though, it's seems like a nice feature.