eriknn/ha-pax_ble #39: The version b394ccf for this integration can not be used with HACS.

JanPetterMG, 2024-02-16:
Trying to update from [a70313e](https://github.com/eriknn/ha-pax_ble/commit/a70313e50b801416716fdba3b9384c8ad695289f) to [b394ccf](https://github.com/eriknn/ha-pax_ble/commit/b394ccf1a1f5ed13519ff797d6448ae36c19de26) was failing for me, and I got this error message:

> The version b394ccf for this integration can not be used with HACS.

I'm not sure what it's all about, but I didn't see any [`hacs.json`](https://hacs.xyz/docs/publish/start#hacsjson) file, so there may be some incompatibility between this library and HACS. The January release of HA each year usually removes support for old legacy features that has been deprecated for a while, not sure about HACS but I wouldn't be surprised if they're following the same norm.

eriknn, 2024-02-16:
I don't know what could cause this. I don't use HACS on this integration myself, but you're free to suggest a fix.

closed 2024-02-16 by eriknn

JanPetterMG, 2024-02-18:
Why is this issue closed?
The readme of this library suggests using HACS, and you're closing this issue as "completed" because you don't use HACS yourself? Pathetic...