eriknn/ha-pax_ble #40: Breaking changes in ba7a70a

JanPetterMG, 2024-02-16:
Downloaded [b394ccf](https://github.com/eriknn/ha-pax_ble/commit/b394ccf1a1f5ed13519ff797d6448ae36c19de26) to replace my existing [a70313e](https://github.com/eriknn/ha-pax_ble/commit/a70313e50b801416716fdba3b9384c8ad695289f) installation has caused trouble, every entity is now duplicated with the suffix `_2` at the end of the entity ID, and every automation and dashboard is now broken.

Not sure what exactly caused this, but after looking at the commits I'm pretty sure there's an breaking change in [ba7a70a](https://github.com/eriknn/ha-pax_ble/commit/ba7a70a8e29324bed2bedc48136765e1260b8723).
This is exactly why we need tagged releases so that we don't need to download untested experimental code...

For anyone reading this, stay away from [ba7a70a](https://github.com/eriknn/ha-pax_ble/commit/ba7a70a8e29324bed2bedc48136765e1260b8723) as well as the newer versions [bef18eb](https://github.com/eriknn/ha-pax_ble/commit/bef18eb83b05487991109e546235092bdb8e3068) and [b394ccf](https://github.com/eriknn/ha-pax_ble/commit/b394ccf1a1f5ed13519ff797d6448ae36c19de26) until this issue has been resolved.

eriknn, 2024-02-16:
Do you have multiple fans?

JanPetterMG, 2024-02-16:
No, just one.

closed 2024-02-16 by eriknn

JanPetterMG, 2024-02-18:
How is this one resolved?
Why is the IDs duplicated, and why not revert? Bug not fixed!

JanPetterMG, 2024-02-18:
Just because I have 1 fan, everyone else in the world should delete and rename 25 entities each, is that a good solution?

If there's a good reason to not revert, then that's perfectly fine, but I'm sure there's a way to make old IDs continue work, while using new IDs for new fans.

eriknn, 2024-02-18:
No, most people have only 1 fan. Including me. I had no issues, but I suspected having more fans could be your issue, as that’s a scenario I can’t test (easily).
As I had no problems with this version, I believe that’s the case for most people. If it happens to be a problem, just remove and add the fan again. It should take you less than 15 seconds.

The cause of this is most likely a fix that makes sure it doesn’t matter if you use upper or lowercase mac address. This might have created a new device with a new identifier in your case. It shouldn’t have, but that’s the only explanation I can see.