eriknn/ha-pax_ble #43: Pax Calima frequently ignores Boost command

Kaaremi, 2024-03-13:
I found this in the logs, is the issue related to the integration or something else, e.g. unstable Bluetooth connection?

This error originated from a custom integration.

Logger: custom_components.pax_ble.coordinator
Source: custom_components/pax_ble/coordinator.py:147
integration: Pax (documentation, issues)
First occurred: March 11, 2024 at 14:44:07 (**19 occurrences**)
Last logged: 18:37:32

Error when writing data: Not connected!

eriknn, 2024-03-13:
You are correct, unstable connection. Happens quite often to me as well. I think this will also be the result if the fan is «occupied» by another connection (phone app etc).

Kaaremi, 2024-03-13:
Thanks,

 

I’ll try to adjust the location of my BT dongle. Hope it helps.

 

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

Kaaremi, 2024-03-18:
Hi,

 

just a small update. 

 

I put the Bluetooth dongle a bit higher with a cable and the simple automation (a door/window sensor triggers the Pax Calima Boost command) seems to work OK, except in the night time.

I have disabled the Pax Calima Silent hours setting which is visible in HA and the Pax Calima’s own Android app. 

I sent a bug report to Pax.se (see attached PDF), but haven’t heard from them so far.

 

Have you encountered this phenomenon? 

 

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

eriknn, 2024-03-18:
This sounds strange… Just remember that the HA-integration does the same as the app - so make sure there’s a problem with the app as well before taking it to Pax.

Kaaremi, 2024-03-19:
Hi,

Last night I did some more testing and found out that the Pax did not start
the boost cycle with a manual command from HA. The switch I used to give
the body command also bounced back to OFF position after few seconds.

After that I launched the Pax Calima Android app and the Boost command
worked fine. However, in the HA the History screen did not show the the rpm
peak.

After 7 o'clock in the morning the HA automation's Boost command worked
fine.

Obviously, no changes to the Bluetooth or any others setups were made.

I hope this helps to analyze the situation.

Parhain terveisin / Best Regards / Mit freundlichen Grüßen

Mikko Kaarela ***@***.***)

eriknn, 2024-03-19:
I find it hard to believe that the clock has anything to do with it. Or maybe you have something creating radio-noise before 7…?

Kaaremi, 2024-03-19:
I have a bunch of other Bluetooth devices, 3 Ruuvi Tags and an Airthings Wave Plus sensor, plus an older Pax Calima. One Ruuvi Tag and the older Pax Calima are downstairs, connected with a Bluetooth Proxy (M5Stack Atom Lite), others communicate via Startech 2567011 EDR Bluetooth Dongle. All other BT devices work fine.

I’ll keep observing and get back to you if something new appears. 

 

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

Kaaremi, 2024-03-19:
Hi,

 

I have a new hypothesis.

 

In the automation I had the condition based on the magnetic door/window sensor closing. Could it be that a short pressing of the WC flush button is not detected by the HA? In any case, now I reversed the condition so then when the flush button is released it triggers the Pax Calima Boost. 

 

Another question is why the boost command given from the HA didn’t work like the Pax Calima app? Maybe the Pax Calima app and the HA Green/ha-pax_ble function somehow differently e.g. in the Boost command’s timing?

 

The relation with the night time could have been just a random thing…

 

I’ll keep you posted.

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

Kaaremi, 2024-03-20:
Hi Erik,

 

I decided to go for a system reset and restored from a full backup. The utterly simple automation (see below) seems to work, i.e. the Bluetooth connection works, but for some reason the automation’s Trace timeline says that the Boost command is an <unknown entity>.

 

Is this normal / irrelevant or indication of a problem?

 

 

alias: Kylpyhuone / WC Tuuletus Boost V2

description: >-

  Kun WC:n ison huuhtelun nappia painetaan niin käynnistetään WC:n tuuletus

  täydellä teholla. Yksinkertaistettu versio ilman Device Offline kontrollia.

trigger:

  - type: opened

    platform: device

    device_id: 5109c471e74ddba035d5fa8a953b9d20                            #Door/window sensor

    entity_id: 506848d35e13b9eb4efdf91347da67f4                         #WC flush button released

    domain: binary_sensor

    id: WC Push

    for:

      hours: 0

      minutes: 0

      seconds: 1

condition: []

action:

  - type: turn_on

    device_id: 82f8a15acb8cd4b9fa07f1aa76c1dea6                         #Pax Calima

    entity_id: 9b49c80f27b9d3fce90fcd005a2dd052                         #Boost command

    domain: switch

mode: single

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

Kaaremi, 2024-04-04:
Hi,

 

I’m not sure if this has anything to do with your integration, but if you have contacts with the Pax Calima manufacturer, maybe you could ask them to have a look at the issue.

Since the Boost command appeared to work in a quite unstable manner I created an automation that uses a Helper Flush#2_Activated that is toggled to TRUE when the magnetic sensor is activated. I also added and UNTIL loop that is repeated with 3 sec. intervals until the Pax Calima confirms that the fan is running in the Boost mode. Besides, I added a counter to find out how many times the Boost command is issued before the Boost starts. 

 

As you can see from the History graph below, it takes quite a few times to get the Boost started.

 

Interestingly, my older Pax Calima in the TV room, acquired 07.04.2021 connected via a Bluetooth Proxy works absolutely fine.

 

 

alias: Bathroom / WC Ventilation Boost V2

description: >-

  When the button for the large flush of the toilet is pressed, the ventilation of the toilet is started

   with full power.

 

   Let's check that the fan definitely goes into Boost mode and if not, repeat

   Boost command until Boost has started. The number of repetitions is sent to Mikon

   mobile phone (test).

trigger:

  - type: not_opened

    platform: device

    device_id: 5109c471e74ddba035d5fa8a953b9d20

    entity_id: 506848d35e13b9eb4efdf91347da67f4

    domain: binary_sensor

    id: WC Push

condition: []

action:

  - service: input_boolean.turn_on

    metadata: {}

    data: {}

    target:

      entity_id: input_boolean.flush_2_activated

  - repeat:

      sequence:

        - type: turn_on

          device_id: 82f8a15acb8cd4b9fa07f1aa76c1dea6

          entity_id: 9b49c80f27b9d3fce90fcd005a2dd052

          domain: switch

        - service: counter.increment

          metadata: {}

          data: {}

          target:

            entity_id: counter.pax_boost_counter

          enabled: true

      until:

        - condition: device

          type: is_on

          device_id: 82f8a15acb8cd4b9fa07f1aa76c1dea6

          entity_id: 9b49c80f27b9d3fce90fcd005a2dd052

          domain: switch

          for:

            hours: 0

            minutes: 0

            seconds: 3

  - service: input_boolean.turn_off

    metadata: {}

    data: {}

    target:

      entity_id: input_boolean.flush_2_activated

  - service: notify.mobile_app_le2113

    data:

      message: >-

        WC:n Boost-command was repeated {{ states('counter.pax_boost_counter') }}

        times at {{ now().strftime('%H:%M') }}

      target: Mikko

    enabled: true

  - service: counter.reset

    metadata: {}

    data: {}

    target:

      entity_id: counter.pax_boost_counter

    enabled: true

mode: single

 

Parhain terveisin / Best Regards / Mit freundlichen Grüßen 

Mikko Kaarela ( ***@***.***> ***@***.***)

eriknn, 2024-04-05:
1. Please respond from Github, or if responding from mail try to remove old text. This thread is getting a bit messy.

2. I am not able to see any of your attachments.

3. If you are unsure if your automation is running you should just check the "last run" timestamp of it. You can also check the trace to see which steps of the automation actually ran.

4. When using the app you will create the connection to the fan first / when you select the fan. If this succeeds, then you have a connection and all other commands from the app should work flawlessly and quickly. So to compare performance with the app, you must compare the initial "connect" command from the app, with the "boost" command from HA.

5. When you activate boost from HA, the "boost"command is sent to the fan. If this command fails (we don't get a connection or reply from the fan) the boost-button will reset to false.

6. Sending a command from HA will take some time - the connection is slow to setup, so 3 seconds between each "send" might be too often.

7. I don't know what the <unknown entity> issue is about, but I would guess that you should try to select the entity again.

Kaaremi, 2024-05-09:
UPDATE
Since I have three Pax Calima fans and only one of them had these Bluetooth issues I asked the manufacturer to replace the motor/electronics unit. The service was excellent and now the fan appears to work with no need to repeat the Boost command.
Case closed.

eriknn, 2024-05-09:
Great to hear! And also a bit weird - must have been very unlucky with that unit!

closed 2024-05-09 by eriknn

Kaaremi, 2024-05-09:
P.S. Is there a way to check the firmware version of Pax Calima?

eriknn, 2024-05-09:
Doesn’t it display under device information? This should be loaded when HA is started. If it failed then you could try to reload the integration.

Kaaremi, 2024-05-09:
Strangely, my oldest Pax Calima (acquired in April 2021) shows

Device info
PAX Calima
by PAX
Firmware: 01.07
Hardware: 01.00

The newer two (March & April 2024)  only show
Device info
PAX Calima

When I click the arrow underneath, I see the big PAX logo and under it,
"1.0.0" without explanation.

Parhain terveisin / Best Regards / Mit freundlichen Grüßen

Mikko Kaarela ***@***.***)

Kaaremi, 2024-05-09:
> Okay, my fan is probably the same as your old fan. They might have changed
> something so that we don’t read the same values. Very strange that the
> values don’t «behave» the same though. All devices should be recreated
> after a HA restart, and they should look the same.

If you'll discuss the matter with Pax, here are the MAC addresses of my
three fans. Maybe this helps to trace the update history...

1. This is the old one from April 2021 and the only one that shows the
firmware version OK.

PaxCalimaTuuletin (58:2b:db:34:b2:be)

2. This I bought in April 2024

Pax AlaWC (58:2b:db:e4:20:02)

3. This I received as a warranty replacement 8th may 2024

Pax Kylpyhuone V2 (58:2b:db:97:e6:a6)

Parhain terveisin / Best Regards / Mit freundlichen Grüßen

Mikko Kaarela ***@***.***)