eriknn/ha-pax_ble #44: *IMPORTANT NOTE*

eriknn, 2024-04-18:
As I have installed balanced ventilation in my home, I no longer need to use my Pax fan. This means that I no longer will be improving the integration. I *might* try to reply and fix any bugs that are reported, but I will no longer be able to test the code.

Hopefully somebody else will take over the development so that new fan models also can be supported in the future!

audunru, 2024-07-28:
It seems to work OK as of august 2024. :-)

el97, 2024-07-28:
No issues yet.

ThomasPontoppidan, 2024-10-10:
Sorry to hear that! I have tried to add my fan but so far no luck. It doesn't auto discover and I can't add it manually either.

eriknn, 2024-10-10:
Check the bluetooth integration / debugging. It should show you any discovered / availiable devices. If it’s not there then you have a problem. If it’s there it might be a range issue.

aridder, 2024-10-13:
Works for me! Thank you for this.

closed 2024-12-13 by eriknn