eriknn/ha-pax_ble #45: Spending most of the time unavailable

jm-cook, 2024-05-27:
Using this integration my fan seems to be unavailable most of the time.

 ```
   This error originated from a custom integration.
    
    Logger: custom_components.pax_ble.coordinator
    Source: custom_components/pax_ble/coordinator.py:255
    integration: Pax Bluetooth (documentation, issues)
    First occurred: 11:09:46 (211 occurrences)
    Last logged: 15:33:40
    
    Error when fetching config data: Not connected!
```

Is there any good strategy for a stable connection and reporting of sensor values?

I have configured scan interval 120 seconds, and fast scan interval 20 seconds.

el97, 2024-05-27:
How do you connect to the fan? Are you using any devices as proxies for ble?

jm-cook, 2024-05-27:
I think it is using the built in bluetooth device directly. This is on a raspberry pi 3

el97, 2024-05-27:
I suggest that you try an ESP32 with the Bluetooth proxy function enabled https://esphome.io/projects/index.html. A smart plug from Shelly would also work https://www.home-assistant.io/integrations/shelly/#bluetooth-support.

jm-cook, 2024-05-31:
I got myself a shelly plug, enabled active bt scanning, proxy, etc, but I don't really understand how this works because nothing changed. Does it automatically just relay messages without any configuration or should it somehow look like a virtual bluetooth device on my rPI?

Guenni9, 2024-05-31:
@el97 and @jm-cook afaik Shelly doesn't support active connections, they only can hear to BLE events. I've also tried this: https://github.com/eriknn/ha-pax_ble/issues/32 without luck..
@el97 did you get it to work finally?

jm-cook, 2024-06-08:
I acquired an esp atom device and installed it as a bt proxy, disabled on board BT, and all systems are now functioning perfectly.

closed 2024-07-17 by eriknn