eriknn/ha-pax_ble #48: Connection Problem Vent-Axia Svara Fan

mattc2, 2024-09-30:
Hi

Thanks for devloping this intergration.

I understand the Svara is basically the same as the Pax and others have sucessfuly configure it. However, I have been trying to add the device via the intergration but it does not want to connect. Any ideas please?  It works fine with the Vent-Axia app.

Rgds

el97, 2024-09-30:
Do you get any log output that says it can see the fan? Some have found it difficult to get a stable connection from where you host your Home Assistant instance to the fan. You can use an ESP 32 device acting as a Bluetooth proxy for a better connection. https://esphome.io/projects/?type=bluetooth

mattc2, 2024-09-30:
Hi

Thanks for your quick response, it is most apreciated. I have identified the problem (finally!) Nothing to do with the intergration or HA generally. The MAC address printed on the device was wrong by one char which I found out by interogating the device with an app.

Thanks again for the intergration, I will be creating some automations tomorow....

closed 2024-09-30 by mattc2

eriknn, 2024-10-01:
Great that it worked out. But ideally this problem shouldn't happen, as HA should discover devices before you even try to add them if I recall correctly...