eriknn/ha-pax_ble #49: How to add fan manually

texnic, 2024-10-01:
Hi there, I have a Vent-Axia Svensa fan. Not sure it will work with this integration, but would like to try. Since my Home Assistant doesn't show my fan automatically, I would like to try "to add it manually through the integration configuration", as mentioned in the "Add device" section. I cannot sort out how to do it. Could you help please?

eriknn, 2024-10-01:
If  click into the integration (after adding it), you should have a «configure» button (this is default for most integrations)

texnic, 2024-10-01:
Response is too slow, I challenge you to do it faster )))

Stupid me, I didn't realize that I have to add the integration again under integrations after installing it in HACS. 
For the record: Go to Settings > Devices & Services > Search for "PAX". If not found, click "Add Integration", search for "PAX". After installing, click PAX > Configure > Add.

Any idea how to find out the MAC address and the PIN code of the fan?

eriknn, 2024-10-01:
I think it’s written on the back of the fan - you have to remove it!

texnic, 2024-10-01:
No luck  :-( Unlike what I read about Svara, Svensa has the propeller on a rotor, and the rotor is part of the motor. On the other hand: how can the Vent-Axia app connect to the fan without me entering the Mac address and the PIN?

eriknn, 2024-10-01:
You could enable debug, that'll tell you connection status. But I think that there are other people as well that has struggled with different versions of the fan. Most likely it's rather easy to figure out - you could scan the device for availiable characteristics. 
Take a look at this:
https://github.com/hbldh/bleak/blob/develop/examples%2Fservice_explorer.py

Then you can compare with the addresses in calima.py
If the addresses are different, but the values are the same('ish), maybe we could seperate the characeristics list in seperate device files.

texnic, 2024-10-02:
I will try what you suggest, thanks! I also opened a thread here: https://community.home-assistant.io/t/vent-axia-svensa-pureair-sense-extractor-fan/777450. Perhaps more people are interested.

eriknn, 2024-10-12:
I'm closing this for now, please open a new issue if you have more information on how we can support your fan

closed 2024-10-12 by eriknn