eriknn/ha-pax_ble #50: Warning in log: async_forward_entry_setups

Guenni9, 2024-10-10:
Dear Erik,
thank you for your great work. I'm using your integration since 1/2 year and it is working very good!

Today, I got following warning:

`Detected code that calls async_forward_entry_setups
 for integration pax_ble with title: 
Pax BLE and entry_id: 748e49f70182ff7302e49d7c6f753338, 
during setup without awaiting async_forward_entry_setups, 
which can cause the setup lock to be released before the setup is done. 
This will stop working in Home Assistant 2025.1. Please report this issue`

Maybe You or  somebody else could advise me, to solve this?
Have a nice day!

eriknn, 2024-10-10:
Okay, probably something that newer versions of HA doesn’t support. Will take a look at it soon!

Guenni9, 2024-10-10:
Thank you very much! FYI, I found similar issues in other integrations, maybe this could help a little?
[e.g. here:](https://github.com/slyoldfox/home-assistant-solaredge-modbus/commit/161387adfa9374643cd0083bca507fb798d77bdd#diff-ef1584470671f19eebfa9b5e3d2cf2fca2133bde65d6f8211778a36ab5a4c176L130)

eriknn, 2024-10-12:
Please try latest release (1.0.1)

Guenni9, 2024-10-13:
Dear Erik,
thank you for your help! I can confirm, the issue is solved!

closed 2024-10-13 by Guenni9