eriknn/ha-pax_ble #54: Issue with two (or more) Pax devices simultaneously?

bfranke1973, 2024-11-28:
Thanks for the great work and support for the Svensa fan!

I am wondering if anyone has come across a problem when using two or more fans simultaneously? I have two fans installed in the house (Pax Calima and Pax Svensa, or more precisely their Vent-Axia counterparts), and I am using two ESP32 based Bluetooth proxies to interface with HA.

Each of the two fans work individually, but as soon as I add the second fan to the Pax integration the one installed before becomes unavailable. I don't think I am running into the 3-connection limitation of the BT proxies as the two fans are the only BT devices I am using. This looks more like an issue with the integration itself.

Has anyone seen this before?

eriknn, 2024-11-28:
Hi! It’s not impossible that I’ve messed something up regarding multiple devices… Can you enable debugging and retrieve some logs?

bfranke1973, 2024-11-28:
Thanks for your quick response. Sure, here is a debug log of when the problem arises:

At this point, a Pax Calima (58:2b:db:28:d5:1e, "Bathroom Fan") is connected, and I am adding a Pax Svensa (4C:96:2D:10:4D:82, "Shower PureSense Fan").

It looks like the connection to both fans can be established, but then Characteristic 7c4adc05-2f33-11e7-93ae-92361f002671 on the Pax Calima cannot be accessed and from there things go wrong.

2024-11-28 22:00:00.444 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 4C:96:2D:10:4D:82
2024-11-28 22:00:00.894 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 4C:96:2D:10:4D:82
2024-11-28 22:00:00.894 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting with pin: XXXXXXXX
2024-11-28 22:00:01.193 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Authorized: True
2024-11-28 22:00:01.397 DEBUG (MainThread) [custom_components.pax_ble] Unloading Pax BLE entry!
2024-11-28 22:00:01.452 DEBUG (MainThread) [custom_components.pax_ble] Setting up configuration for Pax BLE!
2024-11-28 22:00:01.454 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Initializing Calima!
2024-11-28 22:00:01.455 DEBUG (MainThread) [custom_components.pax_ble.coordinator_svensa] Initializing Svensa!
2024-11-28 22:00:01.456 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Bathroom Fan
2024-11-28 22:00:01.456 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Shower PureSense Fan
2024-11-28 22:00:01.460 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-28 22:00:01.460 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-28 22:00:01.460 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 58:2b:db:28:d5:1e
2024-11-28 22:00:01.466 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Bathroom Fan
2024-11-28 22:00:01.467 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Shower PureSense Fan
2024-11-28 22:00:01.483 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-28 22:00:01.483 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-28 22:00:01.483 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 4C:96:2D:10:4D:82
2024-11-28 22:00:01.539 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Bathroom Fan
2024-11-28 22:00:01.539 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Shower PureSense Fan
2024-11-28 22:00:01.556 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Bathroom Fan
2024-11-28 22:00:01.557 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Shower PureSense Fan
2024-11-28 22:00:01.575 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodesecwrite 600
2024-11-28 22:00:01.582 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodespeedwrite 2400
2024-11-28 22:00:01.614 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodesecwrite 600
2024-11-28 22:00:01.630 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodespeedwrite 2400
2024-11-28 22:00:01.633 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Bathroom Fan
2024-11-28 22:00:01.634 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Shower PureSense Fan
2024-11-28 22:00:02.088 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2b:db:28:d5:1e
2024-11-28 22:00:02.215 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 4C:96:2D:10:4D:82
2024-11-28 22:00:02.606 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Bathroom Fan
2024-11-28 22:00:02.607 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when loading config data: esp32-bluetooth-proxy-7dbd00 [78:21:84:7D:BD:00]: Vent-Axia Svara - 58:2B:DB:28:D5:1E: Characteristic 7c4adc05-2f33-11e7-93ae-92361f002671 was not found!
2024-11-28 22:00:02.607 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Reading sensor data
2024-11-28 22:00:02.706 DEBUG (MainThread) [custom_components.pax_ble.devices.calima] Read Fan States: (0, 108, 46, 1254, 2, 0, 2)
2024-11-28 22:00:02.706 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when fetching sensordata: esp32-bluetooth-proxy-7dbd00 [78:21:84:7D:BD:00]: Vent-Axia Svara - 58:2B:DB:28:D5:1E: Characteristic 7c4adc07-2f33-11e7-93ae-92361f002671 was not found!
2024-11-28 22:00:02.707 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Calima: Bathroom Fan data in 1.247 seconds (success: True)
2024-11-28 22:00:03.034 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Shower PureSense Fan
2024-11-28 22:00:03.171 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when loading config data: byte must be in range(0, 256)
2024-11-28 22:00:03.172 DEBUG (MainThread) [custom_components.pax_ble.coordinator_svensa] Reading sensor data
[... cut ...]
2024-11-28 22:00:03.440 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Svensa: Shower PureSense Fan data in 1.957 seconds (success: True)
2024-11-28 22:00:03.444 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.shower_puresense_fan_air_quality (<class 'custom_components.pax_ble.sensor.PaxCalimaSensorEntity'>) is using native unit of measurement 'ppm' which is not a valid unit for the device class ('gas') it is using; expected one of ['m³', 'ft³', 'CCF']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/eriknn/ha-pax/issues

eriknn, 2024-11-28:
Alright, it's at least failing on the characteristics "CHARACTERISTIC_AUTOMATIC_CYCLES" and "CHARACTERISTIC_BOOST".
These are defined in our Svensa integration, as you say you're using. But why is it called Vent-Axia Svara in your logs? I think the Svara is the same as the Calima, have you tried selecting that instead?

bfranke1973, 2024-11-29:
Thanks.

> Edit: I might be misunderstand which models you have here. They are a bit confusing...

Bathroom Fan: Vent-Axia Lo-Carbon, 58:2B:DB:28:D5:1E - which I believe is equivalent to Pax Svara
Shower Fan: Vent-Axia PureAir Sense, 4C:96:2D:10:4D:82 - which I believe is equivalent to Pax Svensa

> Edit2: I see that you're pretty much connecting to both devices at the same time. I have never been able to test with 2 fans. I'm thinking I probably have to read them sequentially...
> Edit3: Making this work sequentially is a little bit of work I think. But we could run a test, to see if this is actually the issue:
> in "init.py" we can put a delay between the instantiation of the coordinators. This should make them out of sync. This would make start up slower, but it should be an okay for testing:

Thanks, I've tried this now, but while this avoids the situation where both devices are being connected at the same time (see log below), the result is still the same: The newly added device is working correctly, but the device that was added before and worked previously, becomes unavailable.

What I can see from the new log is that the new connection is being established successfully, while the previous connection is maintained. Both fans seem to be connected, which is good.

However, I am surprised to see messages relating to [custom_components.pax_ble.coordinator_calima]. This is the device which eventually becomes unavailable, and which was previously set up as Pax Svara. It's almost as if the integration mixes up the type of the previously added device (Calima instead of Svara)! Eventually this fails when one of the characteristics isn't found:  [78:21:84:7D:BD:00]: Vent-Axia Svara - 58:2B:DB:28:D5:1E: Characteristic 7c4adc07-2f33-11e7-93ae-92361f002671 was not found!

Here is my most recent log showing the problem:

2024-11-29 12:50:53.746 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 4C:96:2D:10:4D:82
2024-11-29 12:50:54.343 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 4C:96:2D:10:4D:82
2024-11-29 12:50:54.344 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting with pin: XXXXXXXX
2024-11-29 12:50:54.566 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Authorized: True
2024-11-29 12:50:54.728 DEBUG (MainThread) [custom_components.pax_ble] Unloading Pax BLE entry!
2024-11-29 12:50:54.773 DEBUG (MainThread) [custom_components.pax_ble] Setting up configuration for Pax BLE!
2024-11-29 12:50:55.587 DEBUG (MainThread) [homeassistant.components.http.auth] Authenticated 172.30.32.2 for /api/core/state using bearer token
2024-11-29 12:50:55.594 DEBUG (MainThread) [homeassistant.components.http.auth] Authenticated 172.30.32.2 for /api/services using bearer token
2024-11-29 12:51:04.776 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Initializing Calima!
2024-11-29 12:51:14.779 DEBUG (MainThread) [custom_components.pax_ble.coordinator_svensa] Initializing Svensa!
2024-11-29 12:51:14.780 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Bathroom Fan
2024-11-29 12:51:14.781 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Shower Fan
2024-11-29 12:51:14.782 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-29 12:51:14.782 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-29 12:51:14.782 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 58:2b:db:28:d5:1e
2024-11-29 12:51:14.792 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Bathroom Fan
2024-11-29 12:51:14.792 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Shower Fan
2024-11-29 12:51:14.801 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-29 12:51:14.802 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-29 12:51:14.802 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 4C:96:2D:10:4D:82
2024-11-29 12:51:14.836 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Bathroom Fan
2024-11-29 12:51:14.837 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Shower Fan
2024-11-29 12:51:14.863 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Bathroom Fan
2024-11-29 12:51:14.864 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Shower Fan
2024-11-29 12:51:14.882 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodesecwrite 600
2024-11-29 12:51:14.884 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodespeedwrite 2400
2024-11-29 12:51:14.915 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Bathroom Fan
2024-11-29 12:51:14.917 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Shower Fan
2024-11-29 12:51:15.129 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2b:db:28:d5:1e
2024-11-29 12:51:15.457 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 4C:96:2D:10:4D:82
2024-11-29 12:51:15.812 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Bathroom Fan
[... cut ...]
2024-11-29 12:51:16.460 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Svensa: Shower Fan data in 1.659 seconds (success: True)
2024-11-29 12:51:16.462 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.shower_fan_air_quality (<class 'custom_components.pax_ble.sensor.PaxCalimaSensorEntity'>) is using native unit of measurement 'ppm' which is not a valid unit for the device class ('gas') it is using; expected one of ['ft³', 'm³', 'CCF']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/eriknn/ha-pax/issues

bfranke1973, 2024-11-29:
> If you select "Svara" as device, the integration will use the "Calima"-driver, so that's correct.
> There might be some issue with a shared variable betwen the devices, which should be instance specific... 

Ah,ok. 

> Will try to look at it when I have the time....

Ok, thanks.

eriknn, 2024-11-29:
Aye, the coordinator doesn't start fetching data before it has a listener, so my test didn't work.
In the same init file, you have the definition of a coordinator two places (Under Calima/Svara / Svensa):
Change this:
`coordinator = ....`
To
`
coordinator = ....
coordinator.async_add_listener(lambda: None)
`

eriknn, 2024-11-29:
Okay, but maybe you can show the logs now, as it should be easier to separate the devices…

bfranke1973, 2024-11-29:
2024-11-29 14:27:09.415 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 58:2B:DB:28:D5:1E
2024-11-29 14:27:09.728 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2B:DB:28:D5:1E
2024-11-29 14:27:09.728 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting with pin: XXXXXXXX
2024-11-29 14:27:09.943 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Authorized: True
2024-11-29 14:27:10.070 DEBUG (MainThread) [custom_components.pax_ble] Unloading Pax BLE entry!
2024-11-29 14:27:10.093 DEBUG (MainThread) [custom_components.pax_ble] Setting up configuration for Pax BLE!
2024-11-29 14:27:20.138 DEBUG (MainThread) [custom_components.pax_ble.coordinator_svensa] Initializing Svensa!
2024-11-29 14:27:23.325 DEBUG (MainThread) [homeassistant.components.http.auth] Authenticated 172.30.32.2 for /api/core/state using bearer token
2024-11-29 14:27:30.142 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Initializing Calima!
2024-11-29 14:27:30.143 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Shower Fan
2024-11-29 14:27:30.143 DEBUG (MainThread) [custom_components.pax_ble.time] Starting paxcalima times: Bathroom Fan
2024-11-29 14:27:30.143 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-29 14:27:30.143 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-29 14:27:30.144 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 58:2B:DB:28:D5:1E
2024-11-29 14:27:30.151 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Shower Fan
2024-11-29 14:27:30.152 DEBUG (MainThread) [custom_components.pax_ble.sensor] Starting paxcalima sensors: Bathroom Fan
2024-11-29 14:27:30.153 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Coordinator updating data!!
2024-11-29 14:27:30.153 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Reading device information
2024-11-29 14:27:30.153 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 4C:96:2D:10:4D:82
2024-11-29 14:27:30.196 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Shower Fan
2024-11-29 14:27:30.196 DEBUG (MainThread) [custom_components.pax_ble.switch] Starting paxcalima switches: Bathroom Fan
2024-11-29 14:27:30.216 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Shower Fan
2024-11-29 14:27:30.217 DEBUG (MainThread) [custom_components.pax_ble.number] Starting paxcalima numbers: Bathroom Fan
2024-11-29 14:27:30.225 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodesecwrite 600
2024-11-29 14:27:30.226 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodespeedwrite 2400
2024-11-29 14:27:30.254 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodesecwrite 600
2024-11-29 14:27:30.261 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Set_Data: boostmodespeedwrite 2400
2024-11-29 14:27:30.262 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Shower Fan
2024-11-29 14:27:30.263 DEBUG (MainThread) [custom_components.pax_ble.select] Starting paxcalima selects: Bathroom Fan
2024-11-29 14:27:30.483 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2B:DB:28:D5:1E
2024-11-29 14:27:30.997 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Bathroom Fan
2024-11-29 14:27:31.643 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 4C:96:2D:10:4D:82
2024-11-29 14:27:31.954 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Reading sensor data
2024-11-29 14:27:32.019 DEBUG (MainThread) [custom_components.pax_ble.devices.calima] Read Fan States: (0, 112, 0, 1246, 1, 0, 2)
2024-11-29 14:27:32.224 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Svara: Bathroom Fan data in 2.080 seconds (success: True)
2024-11-29 14:27:32.463 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Updated device data for: Shower Fan
2024-11-29 14:27:32.465 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when loading config data: bluetooth-proxy-upstairs-dd31e4 [F0:08:D1:DD:31:E4]: Pure Sense - 4C:96:2D:10:4D:82: Characteristic f508408a-508b-41c6-aa57-61d1fd0d5c39 was not found!
2024-11-29 14:27:32.466 DEBUG (MainThread) [custom_components.pax_ble.coordinator_svensa] Reading sensor data
2024-11-29 14:27:32.752 DEBUG (MainThread) [custom_components.pax_ble.devices.svensa] Read Fan States: (0, 16, 306, 2030, 2, 760, 238, 7, 0, 24, 0)
2024-11-29 14:27:32.753 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Failed when fetching sensordata: bluetooth-proxy-upstairs-dd31e4 [F0:08:D1:DD:31:E4]: Pure Sense - 4C:96:2D:10:4D:82: Characteristic 118c949c-28c8-4139-b0b3-36657fd055a9 was not found!
2024-11-29 14:27:32.754 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Svensa: Shower Fan data in 2.601 seconds (success: True)

eriknn, 2024-11-29:
This change didn't work either.. .Try this:

```
coordinator = ...
coordinator.async_add_listener(lambda: None)  # Add the empty listener
await coordinator.async_request_refresh()     # Force an immediate update
```

bfranke1973, 2024-11-30:
Success! This seems to work so far. Thank you!

eriknn, 2024-11-30:
Alright, but this makes it a bit easier to see ehat’s going on. Was this the Calima/Svara device, or the Svensa?

bfranke1973, 2024-11-30:
> Alright, but this makes it a bit easier to see ehat’s going on. Was this the Calima/Svara device, or the Svensa?

This was the Svara device, which became unresponsive after adding the Svensa device.

> It’s using the Calima/Svara driver. And the pin seems to be working… 

Yes, the pin appears to be working. In fact, it has been working all the way until the second device got added.

> Did it read all data correct first?

Possibly. The correct data is shown at first, but then not updated any more.

Cheers,
Björn

eriknn, 2024-11-30:
Okay, I have enough info to look into this now. I would guess that when the Svensa device is instantiated, it sets some variables to Svensa-specific values, and this seems to affect the Calima/Svara devices as well. So something seems to be working globally instead of per instance. Not quite sure where that could be, but I'm looking into it :)

bfranke1973, 2024-11-30:
Did you push the updated version? I can't see it yet.

eriknn, 2024-11-30:
No, I just pushed to GitHub. You can download it manually from there, then put in the HA-directory. I don’t want to push a release without testing first.

bfranke1973, 2024-11-30:
No worries, thank you.

Ok, I've done a bit of testing now and it looks your latest fixes indeed solved the issue. Well done and thanks!

Both fans are working, I can control the boost mode, sensor readings are updated and configurations can be changed.

The only very minor thing that I have observed is that for the Pure Sense (Svensa) the mode (under Diagnostic) is shown as "unknown".

Anyway, thanks again for your work and the quick response to the issue!

eriknn, 2024-11-30:
Great stuff! I pushed it as a release now!

Regarding "mode" - all I know is that Svensa has this characteristic, which the Calima also has. I'm not able to test the values on the Svensa though, so now it should show "Unknown: {value}" if the read value is something else than expected. Known values for the Calima are:
0: "MultiMode"
1: "DraftShutterMode"
2: "WallSwitchExtendedRuntimeMode"
3: "WallSwitchNoExtendedRuntimeMode"
4: "HeatDistributionMode"

If you get a value there, it would be nice if you could figure out what it means, or at least tell me the values in a new issue. I'm closing this issue though, as this should be done now :)

closed 2024-11-30 by eriknn