eriknn/ha-pax_ble #55: How to connect a Svensa fan

texnic, 2024-12-14:
Hello, thank you all contributors for implementing Svensa in this extension!  Now I need help connecting it :-) Autodiscovery doesn't seem to work, not sure if I need to initiate it somehow in HA. I see that a couple of BLE scanner apps are able to connect (get some info from Svensa) without me providing any extra input. But this extension is asking for a PIN and I am not sure where to get it. Also, does the MAC-address format matter? 

And if I can somehow debug autodiscovery of Svensa, that would be awesome too.

eriknn, 2024-12-14:
Hi there!
For the Calima you need pin to be able to write, think it’s the same for Svensa.
Autodiscovery only works for the Calima, not sure how to add more names to the list of «supported devices». Regarding mac address, don’t think it’s veru important, but try AA:BB:CC:DD:EE:FF

texnic, 2024-12-15:
> Edit: I mean... really?

Looks different here:
[image]

eriknn, 2024-12-15:
You have a translation issue, seems that no translations are loaded, even the default. Get any errors on load?

texnic, 2024-12-15:
Hm, my language setting is just "English", so I don't expect any translation.

eriknn, 2024-12-15:
Alright, who knows then, maybe it fixed it anyways :/

eriknn, 2024-12-16:
Were you able to add the fan? Would like to close this issue :)

texnic, 2024-12-16:
> Were you able to add the fan? Would like to close this issue :)

No. It took me a while to find the MAC address (apparently, iPhones and Macs are incapable of showing the MAC address, even if they can find and connect the device), had to invite a friend with an Android :-) But now I get "Unknown error occured".

[image]

texnic, 2024-12-16:
I wonder: would it be possible to make PIN optional, if it's only needed to write data? So that, first, the connection can be sorted out, and then the PIN.

eriknn, 2024-12-16:
Oh, are you able to extract logs? I might make the config go to a new screen where you select «Add anyways» or «Go back»…

texnic, 2024-12-16:
Let's try:

[home-assistant_pax_ble_2024-12-16T20-22-04.286Z.log](https://github.com/user-attachments/files/18156706/home-assistant_pax_ble_2024-12-16T20-22-04.286Z.log)

eriknn, 2024-12-16:
Do you have an older version of HA? One of my recent updates leveraged a new functionality in config flow, it looks as though you don’t have this support…
https://developers.home-assistant.io/blog/2024/11/12/options-flow/

eriknn, 2024-12-16:
I've implemented an "add anyways" page locally, but my fan seems to have died to to connection-spamming, and I'm away from home for a couple of days, so I'll have to verify and upload later.

texnic, 2024-12-17:
I updated HA, get a different error now. Log attached.

[image]

[home-assistant_pax_ble_2024-12-17T15-02-50.284Z.log](https://github.com/user-attachments/files/18167388/home-assistant_pax_ble_2024-12-17T15-02-50.284Z.log)

eriknn, 2024-12-17:
Looks like out of range? A lot of people are really struggling with these fans for some reason…

texnic, 2024-12-18:
Hm, trying to debug. My RPi is definitely in the range. But now I see that Vent Axia app also refuses to connect to the fan. I'll contact them. Please keep this ticket open for a while. It'd be nice to collect the known issues and solution here.

eriknn, 2024-12-18:
Have you tried cycling power? The Calima is notorious for this if you mess around with it too much…

texnic, 2024-12-19:
Aha, power cycling helped. Now the connection definitely progresses (it takes a bit of time before I get an error message), and I need to crack the PIN. As described [here](https://github.com/eriknn/ha-pax_ble/issues/51#issuecomment-2425155961), I've sent 00000000 value to 4CAD3...569B2 and got 0x4719a505 back. It is not accepted as PIN by the fan. Tried so far:

[image]

texnic, 2024-12-19:
It works! 

[image]

Shall we put the how-to somewhere?

eriknn, 2024-12-19:
Ah, we encountered that issue when testing, should have warned you 😂 But good job getting it up and running. Should we put all this in the main readme?

texnic, 2024-12-21:
> Should we put all this in the main readme?
> Updated readme.

I was thinking of putting there the complete step-by-step guide. Would you like that? I can write it and share here.

eriknn, 2024-12-21:
Sure, you can edit and create a pull request, or just type it here and I’ll insert it.

texnic, 2024-12-21:
I'll try a pull request, never did it so far, will be a learning. Please bear with me.

eriknn, 2024-12-21:
Awesome! I would suggest using GitHub desktop if you’re a windows user, it simplifies everything!

closed 2024-12-22 by eriknn

eriknn, 2024-12-22:
Thanks for the PR! This pairing sequence should be included in this integration though, so that complx third party app stuff wouldn’t be necessary. Eagerly awaiting your 2nd PR 😂 

In all seriousness, it wouldn’t be too complex to include, but requires quite a bit of changes in the Config Flow. Maybe the initial page could be the same, then after this it could go to pin entry for Pax or pin pairing for your type… Hmm… That’s actually not too hard to implement… *thinking out loud*

eriknn, 2024-12-28:
So... I've implemented all changes needed to Config Flow etc. At least a first draft on how it can be done. All that's missing now is to create a "pair" method in svensa.py. I'm not 100% certain how this works with this fan, so would prefer if somebody else could implement. But if I'm given a clear step-by-step guide on how the process works, I could probably implement it...

texnic, 2024-12-29:
I would be happy to contribute. I don’t have any HA programming experience though. Would you maybe have a hint where to start? I would need to be able to deploy and test the code on my HA before creating a PR.

eriknn, 2024-12-29:
Cool!
I think you should only need to modify this, in svensa.py:
`async def pair(self)`

The characteristic you need to read/write (according to your readme) is already defined as "CHARACTERISTIC_PIN_CODE", and we already have a defined function to write this in the base device:
`async def setAuth(self, pin) -> None`

But we don't have a function to read it. We don't need to define one though, you can just read it directly:
`await self._readUUID(self.chars[CHARACTERISTIC_PIN_CODE])`

So with all this, maybe you just need to keep reading it until it becomes something? Maybe set to zero first? If you create a function (in base_device or in svensa) to read the pin, maybe this could handle the byte swapping etc.
The "pair" function should either return a string with the PIN-code, or throw an exception. This exception will be displayed in the UI. I would expect something like "pairing timed out" or similar to be thrown. You probably need to try reading it for x-number of seconds, so this should give a timeout.

Just shout out if theres something you're wondering about.
Oh, and by the way, I spent way to much time using IDLE to program in python when I started with HA, so if you don't have a proper IDE, you should really get one. That way you'll avoid a million restarts of HA just to fix some silly error that a proper IDE would have told you about - I use VS Code which I think works well both for python and C++ microcontroller programming.

texnic, 2024-12-30:
I cannot say I understand much of the above. But will try. I have PyCharm, so that shouldn’t be a problem. Where I am blocked now is how to set things up to play and debug my code. Do I need to run the dev code on my main HA on RPi? Or can I make my Mac laptop host the HA dev instance? And once the code is ready for PR, should I deploy it on RPi for the final check, or is it unnecessary? Sorry, newb questions, but that’s what I’ll spend most time on I guess. So if you could kick start me, that’d be awesome.

eriknn, 2024-12-30:
Alright,
I'd start with downloading the integration from github - this isn't released to HACS yet, so you need to do this the manual way by downloading and extracting to your components folder. I'm using SMB, and have mapped my HA-installations "\config\custom_components" as a drive on my laptop.
This means that I can open my actual HA-folders directly in my IDE, make modifications, save the files, restart HA and the changes are loaded. I guess there are more professional ways to do this, but I haven't bothered looking much into it. It does make for quite a lot of reboots though, but this has decreased a bit with more experience... ;)

So for some stuff I work like that, and modify and test on my actual HA-instance. Some stuff requires this, like config flow, testing device creation etc. I could probably have run a test setup in a VM or something, but I'm not bothered by some seconds downtime here and there.
But for stuff like you're doing, there are some things that can be tested outside HA. I'd test the functionality "locally" first, with a test-file run from my pc  (using my laptops bluetooth adapter). I uploaded a test file I had lying around for you to start with (not much there, but it's what I have)...

I'd use this and test so that you get the required functionality - that is, you're able to read the PIN code. Then you can put it into the actual integration and test the same from there.

texnic, 2025-01-06:
Progress so far: Still uncertain about how to set up the dev environment. https://developers.home-assistant.io/docs/development_environment/ mentions explicitly that I cannot use Bluetooth on Mac if using containers, so I cannot use their guide. I also don't feel to comfortable changing the code directly on my HA, since it controls quite a bit at my home, and I don't want it to crash. Hence, I am now trying to instal HA directly on my Mac and see if I can run it to develop the pairing functionality. To be continued...

eriknn, 2025-01-06:
Yeah, I see there could be some issues. Maybe just run «regular» python on your machine to test the specific connection logic, then I can put it in the integration? We just need to know what to write/read, and when this is «success». Based on your docs I would guess we can just read the PIN-datapoint until it becomes something other than 0, and that’s it…

texnic, 2025-01-27:
Hi Erik,

I don't think I'll be able to create a code that can be directly integrated. My async Python knowledge is nil. But maybe you manage to make some sense of what I've achieved? The following will find all Svensa devices around and get the PIN for the one put in pairing mode. By the way, it would be nice to get rid of the MAC address input field, as Apple users don't have easy way to find those. The code below works on Macbook, and doesn't rely on MAC addresses.

```
import asyncio
from bleak import BleakClient, BleakScanner

CHARACTERISTIC_UUID = '4cad343a-209a-40b7-b911-4d9b3df569b2'

async def main():
    devices = await BleakScanner.discover(timeout=2)
    while True:
        for d in devices:
            if d.name == 'Svensa':
                print("Svensa found")
                async with BleakClient(d) as client:
                    if client.is_connected:
                        print(f"Connected to Svensa at {d.address}")

                        # Write data to the characteristic
                        await client.write_gatt_char(CHARACTERISTIC_UUID, bytes([0x00, 0x00, 0x00, 0x00]))
                        print("Zeros written")

                        # Read response from the characteristic
                        response = await client.read_gatt_char(CHARACTERISTIC_UUID)
                        # Convert to Svensa pin
                        pin = int.from_bytes(response[::-1])
                        print(f"PIN: {pin}\n")

                        if pin:
                            return
                    else:
                        print("Failed to connect to the device.")

asyncio.run(main())
```

PR #57 (2024-12-22): Instructions for Svensa fan
Added step-by-step instructions for Svensa fan. Resolves #55.