eriknn/ha-pax_ble #56: Question: How to find this integration in HACS?

BoneheadFraggle, 2024-12-14:
I tried to search for PAX, Calima, ha-pax_ble and all other combinations I could think of, but found nothing.
Manual copy worked fine, but since it says HACS installation is available, it would be nice to find it :-)

eriknn, 2024-12-14:
I think you need to add the repository manually in HACS?

BoneheadFraggle, 2024-12-15:
> add the repository manually in HACS

Ah, ok. No worries then. I thought it was already in the repository and that I just couldn't find it.
Suggestion: Clarify that in the install instructions, maybe even with a link to HACS guide: https://www.hacs.xyz/docs/faq/custom_repositories/

The integration works perfectly with my Calima's. Thanks for your efforts!

eriknn, 2024-12-16:
Yeah, I'm going to try to add it to the official HACS repository :)

eriknn, 2024-12-17:
PR created in HACS :)

closed 2024-12-17 by eriknn