eriknn/ha-pax_ble #58: Issues connecting to Svara

chinery, 2025-01-13:
Hi there,

My Svara has been working perfectly with this integration for weeks and it's great, so firstly thanks for the hard work! 

Sadly at some point it went offline where all the attributes just read "unavailable" (possibly timed with a HA update/restart, I'm not sure). I could still connect with the Vent Axia app no problem.

After trying various disable/reenable cycles to try to get it to reconnect and nothing worked, I eventually removed the fan from the integration and tried to re-add it. When I hit submit I get a red "Unknown error occurred" message at the top of the pop up. I tried exporting the logs, and I see the error below, any ideas what I can try?

```python
2025-01-13 20:18:01.807 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting to 58:2B:DB:35:BE:3E
2025-01-13 20:18:01.811 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2B:DB:35:BE:3E
2025-01-13 20:18:01.811 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connecting with pin: 08434299
2025-01-13 20:18:01.812 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 285, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 341, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 388, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 491, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pax_ble/config_flow.py", line 226, in async_step_add_device
    await fan.setAuth(user_input[CONF_PIN])
  File "/config/custom_components/pax_ble/devices/base_device.py", line 137, in setAuth
    await self._writeUUID(self.chars[CHARACTERISTIC_PIN_CODE], pack("<I", int(pin)))
  File "/config/custom_components/pax_ble/devices/base_device.py", line 103, in _writeUUID
    await self._dev.write_gatt_char(char_specifier=uuid, data=val, response=True)
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 786, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/client.py", line 862, in write_gatt_char
    assert_reply(reply)
    ~~~~~~~~~~~~^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] No ATT transport
```

eriknn, 2025-01-13:
Hi there, and thanks for kind words!

Unfortunately this error is new to me, but I would suspect an issue with your bluetooth device/driver. Do you have any other bluetooth devices in your integration? Tried enabling/disabling passive scanning on the adapter?

eriknn, 2025-01-14:
When you had those «removal issues», did ypu restart quickly after removing it? I’ve encountered the same before, and it’s most likely an error on my side, not making sure the change is saved after removal. If you wait longer I think the config will automatically saved anyways. Will take a note of it and look at it next time I’m making changes.

Have you cycled power on the fan?

chinery, 2025-01-14:
I just double checked that bluetooth proxy setting on my NSPanel – I didn't think it was working, but I tried disabling my Intel adaptor in the bluetooth integration, and there were still bluetooth entries for the everything presence lite showing up in the debug logs. So I tried to add the Svara and it connected again!

I'll let you know if anything changes but for now seems to be up and running. Thanks for the help!

closed 2025-01-14 by chinery