eriknn/ha-pax_ble #62: Issue with controlling the Pax Calima Fans

Pheomatar, 2025-02-10:
Hello I need help as I cannot find the answer anywhere.
I have Pax Calima integration from Hacs for 3 Pax Calima Fans.
I also have an esp32 generic wroom 2 that I uploaded with a bluetooth proxy from esphome in which I added a button for restart in esphome and some sensors for the board.
My problem is that the first time it worked almost flawless to connect to the fans but now for some reason it acts up. The reason I added the restart button the esp is so I can have access again.
I try to press the boost button for example and it does not work. It show the info on it like it should work but it does not. the only way I can access it again is if I restart the esp.
I tried 3 esp boards all act the same. I tried messing with the bluetooth proxy, still the same I do not have any idea how do solve it if someone can help.
I want to be able to make automation for example I have an aqara sensor in the bathroom and when it rises over 50% both of the fans to start on boost mode or in the toilet I have a presence sensor if someone is there for the boost button to start. I am not able to do that because they do not respond properly.

eriknn, 2025-02-10:
I have no experience using BT proxies, so unfortunately I can’t help you with this :( Hopefully someone else knows more…

Pheomatar, 2025-02-10:
Ok. Is there any other way that I can control them in Home Assistant? without using a BT proxy?

el97, 2025-02-10:
I need some clarification here.

> My problem is that the first time it worked almost flawless to connect to the fans but now for some reason it acts up. The reason I added the restart button the esp is so I can have access again.

Do you mean that you added a restart button in the YAML configuration for the BT-proxy?

Pheomatar, 2025-02-10:
Yes. It works almost ok when I plug it in for a few minutes. 1-2 cycles of boost mode and then I have to restart it again.

el97, 2025-02-10:
Based on the information provided here, the BLE proxy has its limitations with how many connections you can have to a single ESP device at the same time. Perhaps you can use a second ESP device at the same time to make the connection more redundant.
https://esphome.io/components/bluetooth_proxy

Pheomatar, 2025-02-10:
thank you that is a good idea and I will try. Also do you have any idea if I can filter the fans maybe? so it only look for those?

el97, 2025-02-10:
You should be able to filter out how the fans connect based on the MAC address. Unfortunately, I don't know how to bind a specific device to a BLE proxy.

Pheomatar, 2025-04-19:
Thank you. Meanwhile I solved it by changing the generic Bluetooth proxy esp32 with esp32 C6. It was reliable just a couple of times I had tu turn on and off the fans. 
I will also test how the previous Esp32 will work with the new code.

PR #66 (2025-04-17): pax_ble: Improve BLE connection resilience
Reuse a single BleakClient instance in BaseDevice.connect, instead of recreating a new client on every connect call.

Introduce BaseDevice._with_disconnect_on_error to wrap all GATT read/write operations. On any exception, the client will be cleanly disconnected to prevent stale sessions.

Enhance BaseCoordinator._safe_connect with a fixed five‑attempt exponential backoff (1s, 2s, 4s, 8s), removing the previous retry parameters and making retry logic consistent across all coordinators.

Fixes #62.