eriknn/ha-pax_ble #63: Temperature and Humidity calculations

toast1234, 2025-02-23:
Just to share my experience.
I have 5 Vent-Asia and 2 Pax Colima. One of each brand I have monitored more closely in temperature -1C to 25C and Humidity 20%-100%
I use this formula that seems to work quite well compared to a DHT22 sensor.
Humidity get less resolution and accuracy below 35 but otherwise follow the DHT22 quite well. From time to time there is a +/- 5% diff in humidity but in general it lines up over time.

        return FanState(
            round(math.log2(v[0])*10 + 10, 2) if v[0] > 1 and v[0] < 512 else 100 if v[0] >= 512 else 20,
            round(math.log2(v[1])*10 - 45, 2),
            v[2],
            v[3],
            trigger,
        )

On some fans the temperature seams a bit higher and I use an offset of -47 rather than -45

For fans that have a bit weaker Bluetooth signal the read might fail quite often from time to time. Can this be improved in the code by doing a retry loop of say 5 reads if initial connection immediately fails?  Sometimes it also looks like the temperature is updated but not the humidity.

eriknn, 2025-02-23:
Alright, I just plotted your functions vs the existing ones, and they do differ a bit, especially the humidity function. I think your values would have given me a reading > 100% when I was using this fan. Maybe somebody else with a secondary sensor could test your values and verify?

toast1234, 2025-02-24:
Yes I also noticed that so I put in that anything over 512 gives 100%. Its close enough for me.

eriknn, 2025-02-24:
Yeah, I guess. Let's get some more feedback before we change in the lintegration.

Regarding temp updating and not humidity: This shouldn't be possible, they're both read in the same bluetooth telegram! If you really think this is the case, you must provide so logs so that we can look into it!

elgeniskogen, 2025-05-18:
I am getting really weird  readings for humidity on my Pax Calima. It is like it needs to be triggered by REALLY high humidity before it starts reporting. Are you seeing the same things or do you have a nice curve? The connection should be good as I have a Bluetooth proxy nearby. 

[image]

eriknn, 2025-05-18:
Yeah I had something like that. It’s more like an on/off sensor really.

closed 2025-08-19 by eriknn