eriknn/ha-pax_ble #64: PIN

Itsmmai, 2025-04-13:
Hi,

I.m so sorry but I can't connect te Svensa Fan (I have the Vent Axia). it goes wrong with retrieving a PIN code. Maybe it's because I'm not native English but I don't get it. Can you please explain me how to retrieve the PIN ?

Kind kind regards for your help!

Ab

(I know how to run BLE and write to the device, not sure what to write and what I should get from the device)

eriknn, 2025-04-13:
Are you using the latest beta, 1.1.10b?

Itsmmai, 2025-04-13:
No, do I need to do so?

eriknn, 2025-04-13:
See my updated post.

Itsmmai, 2025-04-13:
> See my updated post.

Sorry, I do not see any changes? Do I let pin code empty now?

eriknn, 2025-04-13:
You need to enter the correct mac. Pin empty I think.

Itsmmai, 2025-04-13:
> You need to enter the correct mac. If entered PIN (can be empty) is rejected, you’ll get the option to accept the wrong pin or try to fetch from device.

{"name":"Svensa","address":"4C:96:2D:62:66:DE","rssi":-94,"manufacturer_data":{},"service_data":{},"service_uuids":["0000180a-0000-1000-8000-00805f9b34fb"],"source":"8C:90:2D:41:E4:A2","connectable":true,"time":1744576364.4526176,"tx_power":3}

Address is 4c:96:2D:62:66:de or am I is that a mistake?

eriknn, 2025-04-13:
Yeah looks correct

Itsmmai, 2025-04-13:
still nothing :-( Is there a log I can check, or can send you ?

eriknn, 2025-04-13:
Hou can enable debuggong under the integration, that’ll give you logs. I suppose it’ll just say «failed to connect» but worth a try… The info you pulled with mac address, was that from HASS? If so it should be able to connect…

Itsmmai, 2025-04-13:
Do I need to enable Bluetooth  pairing while adding to HA?

eriknn, 2025-04-13:
I guess, but probably not before you try to fetch the PIN. Din’t really know these fans…

Itsmmai, 2025-04-14:
It is working now. Thanks so far, I think it was a transmit/receive thing. Can I now also send data to the Fan ? for example can I send a pause command ?

eriknn, 2025-04-14:
Great stuff. Does this mean that the PIN-retrieval worked?
You should have lots of entities for controlling the fan. There's no "pause", but you can probably change the mode to "off" or something. I don't quite remember what entities this fan will give you...

Itsmmai, 2025-04-14:
Yes it worked great! I will find out. Thanks a lot!

eriknn, 2025-04-14:
Superb! That means that I can release this version as no longer beta, thanks for the feedback :)

Itsmmai, 2025-04-14:
> Superb! That means that I can release this version as no longer beta, thanks for the feedback :)

I will try it again tomorrow (with a fresh install) and will let you know (double test)

closed 2025-04-14 by eriknn