eriknn/ha-pax_ble #67: Svensa PIN retrieval vs. integration UI

texnic, 2025-04-25:
Hi Erik, I see PIN retrieval has been implemented for Svensa in version 1.1.10. I've been trying to test it, but got stuck at the first screen where I am still asked to enter the Bluetooth address and the PIN code. How should I initiate PIN retrieval?

[image]

eriknn, 2025-04-25:
Leave it blank or enter whatever. It will then fail and ask iif it should retrieve.

texnic, 2025-04-25:
I guess we need to update the readme :) What I tried:

- Leave MAC and PIN fields empty. The integration complains not all required fields are filled.
- Enter "whatever" MAC and PIN. The integration complains "Failed to connect".
- Enter correct MAC and wrong PIN. The integration proceeds to the next screen, where it offers options to "Accept wrong PIN", "Try new PIN", "Start pairing". I chose the last one, put the fan in pairing mode and pressed submit. After a while, the integration showed "Failed to retrieve a valid PIN after 5 attempts". 

Does the integration require MAC and PIN now? Or should it find a device put in pairing mode? If the latter, at which step is pairing initiated (by the integration)? If the former, I wonder if it would be possible to get rid of MAC altogether (I posted the code that allows it earlier, https://github.com/eriknn/ha-pax_ble/issues/55#issuecomment-2617045718).

eriknn, 2025-04-25:
Your experience is correct. Mac still has to be typed - everyone with HA can find this under the bluetooth integration. As it is now - after you press submit it will try to reach the mac you entered and ask for PIN.

I’m sure a discovery etc could be added if anyone can make a PR. I don’t have this fan, so I don’t want to do much about it. A possibility is also to have the Svensa devices autodiscovered like the PAX, but I’m not sure how to ads more fans to this (in manifest)

texnic, 2025-04-26:
I recognize the lack of strong motivation to make Svensa work, and appreciate the work you've already done. But I am not sure what you mean by "correct experience". I cannot make the integration retrieve the PIN. 

I have now tested with the correct PIN, and got the following experience. After entering the correct MAC and PIN, the integration tells me the PIN is incorrect ("Pax: Wrong PIN" window). It doesn't matter if I put the fan in pairing mode or not. If I then select "Start pairing" while the fan is in the pairing mode, it says "Failed to retrieve a valid PIN after 5 attempts". If I select "Accept wrong pin", it returns, after a while, "Device XX successfully added" (I don't have to put the fan in pairing mode in this case). After restarting HA, the fan appears in the list. However, I cannot control it. So apparently you implemented the read-only option (it would be nice to make it explicit, rather than call it "accept the wrong PIN"). But I cannot make the fan controllable anymore. I cannot make PIN retrieval work, and the correct PIN found with the way I discovered earlier doesn't seem to work anymore.

> everyone with HA can find this under the bluetooth integration

Could you clarify what it means please? My Bluetooth integration only shows RPi Bluetooth adapter. So it doesn't help finding the MAC address of the Bluetooth adapter of the fan, AFAIK.

eriknn, 2025-04-26:
By «correct experience» I mean that it is what is expected from the integration as it is now.

The PIN you retrieved with your code is probably byteswapped, so you need to fix that before entering is this integration.

Under your BT adapter you can download diagnostica which show you all discovered devices.

The PIN-retrieval in this integration might work, I think one person got it working. But I’m not able to test anything as long as I don’t have the fan, and that’s why I don’t want anything to do with it. I added a framework for it which should make it easy for anyone else to test and improve the code.

I agree that the process for adding the device with/without PIN etc is rather messy now, and I can fix that if someone comes with clear suggestions on how it should be.

texnic, 2025-04-27:
I agree with what you are saying.

> The PIN you retrieved with your code is probably byteswapped, so you need to fix that before entering in this integration.

Let's sort this out. 

The complete process to obtain the PIN to enter in the integration is currently described in svensa.md:

- Obtain the bytes
- Flip the bytes of the PIN (dd:cc:bb:aa)
- Convert this value to decimal
- Use the result in the integration PIN field

This is what my code (published in the [other ticket](https://github.com/eriknn/ha-pax_ble/issues/55#issuecomment-2617045718)) does. It used to work (that's why it's in the documentation for the integration). What does the integration expect now? Bytes before swapping? Bytes converted to decimal before swapping? 

At the very least, I'd like to update the documentation, I can manage that much. I would either put my code from the ticket into svensa.md, or, preferred, understand how to make the integration work and update the documentation with that.

FWIW I started learning async Python to be able to dive into your code. Then hopefully I can contribute more than just editing md-files.

> The PIN-retrieval in this integration might work, I think one person got it working.

I saw that chat, and it puzzles me. I need to do more testing to understand. But for starters, I need to make integration work at least as it did before, otherwise the regression is too large.

eriknn, 2025-04-27:
Yeah that procedure should be correct - byteswap and convert to decimal.

Async is pretty much the same as regular code, so that shouldn’t be too difficult to learn.

texnic, 2025-04-28:
> Yeah that procedure should be correct - byteswap and convert to decimal. Nothing has changed!

I will try again.

eriknn, 2025-08-19:
Closing this, open a new thread for follow up on PIN handling