eriknn/ha-pax_ble #68: not valid repository?

trucktire, 2025-05-15:
Hi. I'm new to homeassistant, so I got pretty much no idea what I'm doing. I'm coming from Samsung Smartthings, and is an avid windows user, so it should say something about my mad skillz!?

Anyway, I'm having a bit of a problem understanding your instructions? I'm trying to add your github link to repo manually but I'm getting:

"https://github.com/eriknn/ha-pax_ble.git is not a valid add-on repository"

Could you tell me what I'm doing wrong please?

Thanks.

trucktire, 2025-05-15:
Nevermind. Som further searching solved it for me. Patience is golden...

closed 2025-05-15 by trucktire

eriknn, 2025-05-16:
Great stuff! I think it got accepted to HACS, so it should just be searchable now!