eriknn/ha-pax_ble #69: getHumidity

cybermatte, 2025-06-06:
Hi,

I have a Pax calima and in my integration I can not read the humidity value, could it be that the functions 
getHumidity
setHumidity

Are missing in calima.py ?

/Mattias

eriknn, 2025-06-06:
I don’t understand what you mean, can you give more details?
Are you using the integration in HA or for something else?

cybermatte, 2025-06-07:
So, I'm using Home Assistant and HACS with your integration to Pax.
I do have the Calima fan and I have it setup from the the app in the beginning.
I have been using your integration for 1-2 years now.
So far the fan it self is configured to react on high humidity and does so after some time.
However today I have to use an external humidity sensor to read when the humidity goes above 65% in the bathroom and then trigger a Boost in the integration.
What I am missing is the humidity value from the integration.

Sadly the Calima humidity is always 0 and as to your documentation it is supposed to be 0 at all time unless humidity is high.
But I have never seen the humidity to move above 0 at any given time.
What is high? I think that after a shower and my other sensor shows 70% that is high.

Maybe the issues is that the function getHumidity is not implemented in calima.py?
It is implemented in svensa.py.

cybermatte, 2025-06-08:
Before shower it reads:
2025-06-08 08:04:28.412 DEBUG (MainThread) [custom_components.pax_ble.devices.base_device] Connected to 58:2b:db:00:d3:bd on attempt 2
2025-06-08 08:04:28.511 DEBUG (MainThread) [custom_components.pax_ble.devices.calima] Read Fan States: (3, 86, 6, 840, 1, 0, 0)
2025-06-08 08:04:28.721 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Calima: Pax data in 15.274 seconds (success: True)

After shower and when I have triggered a boost:
2025-06-08 11:14:06.447 DEBUG (MainThread) [custom_components.pax_ble.coordinator_calima] Reading sensor data
2025-06-08 11:14:06.597 DEBUG (MainThread) [custom_components.pax_ble.devices.calima] Read Fan States: (13, 107, 38, 2151, 17, 0, 0)
2025-06-08 11:14:06.701 DEBUG (MainThread) [custom_components.pax_ble.coordinator] Finished fetching Calima: Pax data in 0.254 seconds (success: True)

In the display humidity is always 0.
When I triggered the boost manually my external humidity sensor was at 97%.

eriknn, 2025-06-08:
I’ll take a quick look at the code later today, and check with your debug values!

eriknn, 2025-06-10:
Sorry, it took some time before I got to this:
The humidity is read in calima.py, in the function "getState". This return temperature, humidty+++, and debugs the values that you got from your log. In this order:
Hum   Temp  Light FanSpeed Mode Tbd   Tbd

So your humidity sensor value goes from 3 -> 13.

Before we return this as a %-value, we recalculate the humididy value:
humidity = round(math.log2(v[0] - 30) * 10, 2) if v[0] > 30 else 0,

Your value is > 30, so it will always return 0.

Either  you have different hardware/software in your fan, so it doesn't actually send the humidity, or the sensor is broken.

Are you sure that the air around the sensor is replaced with humid air? Is it running in trickle mode?

closed 2025-06-24 by eriknn