eriknn/ha-pax_ble #78: Sensor information showing Unknown

joerichardson69, 2025-08-13:
I had 1 Svara fan configured for a few months and it has been working fine. Today I have tried to add more fans. At first from the UI and then manually in the JSON in /config/.storage/core.config_entries. What ever I try I can not get the sensor information to show. I tried deleting and removing the integration, which has made it worse as now I can not even find the original fan from the UI. 

- I am hoping someone can confirm the below JSON is correct.
- Also any guidance on debugging the UI to find out why it wont let me add multiple fans, I have 5 to add.

{
  "created_at": "2025-05-20T12:40:42.293871+00:00",
  "data": {
    "devices": {
      "58:2B:DB:96:0E:0E": {
        "mac": "58:2B:DB:96:0E:0E",
        "model": "Svara",
        "name": "Fan1",
        "pin": "3372xxxx",
        "scan_interval": 300,
        "scan_interval_fast": 5
      },
      "58:2B:DB:34:DF:C5": {
        "mac": "58:2B:DB:34:DF:C5",
        "model": "Svara",
        "name": "Fan2",
        "pin": "8144xxxx",
        "scan_interval": 300,
        "scan_interval_fast": 5
      }
    }
  },
  "disabled_by": null,
  "discovery_keys": {},
  "domain": "pax_ble",
  "entry_id": "01JVPVBRHN1HW6X92AWDCFZXA1",
  "minor_version": 1,
  "modified_at": "2025-05-20T12:40:42.293883+00:00",
  "options": {},
  "pref_disable_new_entities": false,
  "pref_disable_polling": false,
  "source": "user",
  "subentries": [],
  "title": "Pax BLE",
  "unique_id": "Pax BLE",
  "version": 2
}

eriknn, 2025-08-13:
Hi there!
First, just enable debugging for the integration, and that should give you some info in the logs. Think that logs UI issues as well.

Unfortunately, I only have 1 fan, and so I haven’t tested multiple fans at all. Adding config entries manually sounds like a bad idea - there should only be one instance of the pax integration anyways! Multiple fans will reside in the same instance!

joerichardson69, 2025-08-14:
I have worked out what is wrong. The shelly proxy does not support two way bluetooth traffic, it only listens and relays. I am going to try some M5Stack ATOM Lite ESP32 Development Kits which hopefully will solve the issues.

eriknn, 2025-08-14:
Thanks for the update! Maybe this should be included in the readme 🤔